ra3xdh/qucs_s #1: Heads up:  Upcoming Xyce deployment changes

tvrusso, 2017-09-26:
The Xyce team has decided to do away with the "runxyce" and "xmpirun" scripts it has been distributing with its release binaries.  You might want to update Qucs-s so that it does not default to using these scripts to run Xyce.

The "runxyce" script did essentially nothing but take all the command line arguments it was given and pass them to the Xyce binary that lives in the same directory as the script.  As such, it was pretty much a pointless vestige of very old releases that were not built with "-rpath" so they could find their shared libraries.

"xmpirun" was similarly a vestige of when Xyce needed "LD_LIBRARY_PATH" set in order to run, but clumsily attempted to tease apart its command line arguments to determine which were to be passed to "mpirun" and which to Xyce.  This script was tedious to maintain, and could not handle some of the more esoteric options to mpirun.

Since neither script is particularly useful anymore, and both are maintenance headaches, as of Xyce release 6.8, these scripts will no longer be included with Xyce release binaries, and the correct way to run Xyce in serial will simply be to invoke it as "/path/Xyce" instead of "path/runxyce."  In parallel, one will have to run it as "mpirun -np <n> [mpioptions] /path/Xyce [xyceoptions]" instead of "/path/xmpirun -np <n> [mpi and xyce options]".

It has always been possible to to run Xyce this way through Qucs-s by editing the default simulator settings.  But after release 6.8, the defaults won't be usable.  You might want to change them now, since the new way of running will work with older builds at least back to release 6.6 (when we started releasing binaries with RPATH set correctly).

Thank you for your continued work in making Qucs-s usable with Xyce.  Qucs-s is an excellent choice as a schematic capture tool for use with Xyce, and we are encouraging our users to make use of it.

ra3xdh, 2017-09-27:
Thank you for sharing this important information. Yes, I will update defaults as soon as Xyce-6.8 will be released. Ideal solution will be to recognize Xyce version and suggest proper defaults. I will probably try to add an option "Find available simulators" in the "Simulator settings" dialog. Also Qucs-S could request to automatically scan system for available simulators at the first start.

ra3xdh, 2017-10-25:
I have just updated default paths for Xyce. The `runxyce` script was removed from paths. Look at https://github.com/ra3xdh/qucs_s/commit/c45bc09437b9f47af4bf0f9a598c4b85787e7288

tvrusso, 2017-10-25:
Great.  Thank you.  I would like to try out the new qucs_s stuff on the current branch, but have not built qucs_s since you created this new repo.

I just tried building qucs_s by following the instructions in README.md (and since I'm building current, I'm leaving off "-DWITH_SPICE=ON"), and am getting a compile error.  Here's what I tried:

    git clone <qucs_s repo>
    cd qucs_s
    git checkout current
    mkdir builddir
    cd builddir
    cmake .. -DCMAKE_INSTALL_PREFIX=<my install path>
    make

This gets me:  

    [  0%] Generating moc_componentdialog.cxx
    [  0%] Generating moc_spicedialog.cxx
    [  0%] Generating moc_optimizedialog.cxx
    [  0%] Generating moc_spicefile.cxx
    Scanning dependencies of target components
    [  1%] Building CXX object qucs/components/CMakeFiles/components.dir/DLS_1ton.cpp.o
    In file included from /users/tvrusso/src/qucs_s/qucs/components/DLS_1ton.cpp:11:
    In file included from /users/tvrusso/src/qucs_s/qucs/components/DLS_1ton.h:14:
    /users/tvrusso/src/qucs_s/qucs/components/component.h:23:10: fatal error: 
          'element.h' file not found
    #include "element.h"
                  ^
    1 error generated.

Can you let me know what I'm doing wrong?

tvrusso, 2017-10-25:
I just tried creating the build dir under qucs_s/qucs (the way the instructions on https://ra3xdh.github.io/ said to do for the old repo) instead of under qucs_s (the "top level" directory from the clone I thought that README.md referred to), and *that* seems to work.  Or at least, I don't get that compile error.

tvrusso, 2017-10-25:
OK, when I modified my build procedure to be:

    git clone <qucs_s repo>
    cd qucs_s
    git checkout current
    cd qucs
    mkdir builddir
    cd builddir
    cmake .. -DCMAKE_INSTALL_PREFIX=<my install path>
    make
    make install

it works fine and creates the qucs-s executable (only).  I can also build qucs-activefilter and its friends separately by making a build directory in that subdirectory and doing another cmake/make.

But if I make the build directory directly under qucs_s (which I presume should build qucs-s and all the other programs)  the result is that failure to find "element.h"

ra3xdh, 2017-10-25:
The build issues are very strange. I will try to check the clean build
tomorrow morning. Probably there are some side-effects of auto-merging.

<https://github.com/ra3xdh/qucs_s/issues/1#issuecomment-339385042>, or mute
the thread

tvrusso, 2017-10-25:
Thank you, Vadim.

ra3xdh, 2017-10-26:
Yes, I can reproduce this issue. I have only found that it appears after the https://github.com/ra3xdh/qucs_s/commit/9a1a7b17ec71dfd6f32e6f2e08bb8f5577084184 , but I cannot found why.

ra3xdh, 2017-10-26:
It become clearer now. Something went wrong with qucs_SOURCE_DIR variable. I will try to fix it today or tomorrow.

ra3xdh, 2017-10-26:
I have fixed this problem. Look at the commit https://github.com/ra3xdh/qucs_s/commits/current on the `current` branch.

tvrusso, 2017-10-26:
Nice.  Thank you for the fix.  I've got it all built now.

Looking forward to checking out some of the other additions you've made related to Xyce (sensitivity and .step work).

closed 2017-10-27 by ra3xdh