ra3xdh/qucs_s #2: W(CSW) is not working

cvgaviao, 2017-09-27:
Tried to use the spice switch W in a small circuit. And couldn't find a way to make it to work with both ngspice and xyce.

I used the format that is indicated in the Ngspice manual:

> It=0.01 Ih=0 RON=100Ohm ROFF=10MOhm

But both xyce and ngspice complains about it.

[image]
[image]
[image]

[Relay.txt](https://github.com/ra3xdh/qucs_s/files/1336558/Relay.txt)

ra3xdh, 2017-09-27:
@cvgaviao , It is not a bug. There are indeed missing parameters. Current controlled switches require definition of controlling source and modelcard (`.MODEL`). Look the switch definition example at the Ngspice manual:
~~~
w1 40 0 vm3 wswitch1 off
.model wswitch1 csw it =1m ih = 0.2m ron =1 roff =10k
~~~

ra3xdh, 2017-09-27:
Also SPICE notation should be used for SPICE-only devices. For example SPICE doesn't support measurement units (like Ohm) and has a different prefix system (`Meg` instead of `M`)

tvrusso, 2017-09-27:
Several things here:

1. You are not specifying any voltage source whose current should control the switch.  Both ngspice and Xyce require that.  
  1a.  Qucs-s doesn't seem to make it obvious that this is required, and should probably have a slot in its device parameters that clearly indicates that it should have a current probe name specified.
2. You are specifying all of the parameters of the device as instance parameters in the W1 device.  Neither ngspice nor Xyce support that for the W device (at least for the version of ngspice I have installed).  The various parameters are supposed to go in a .model and the name of that model given for the W device.
3. Xyce's switch devices are not the same as ngspice's, and take different model parameters.  For example, there is no It and Ih parameter for Xyce's devices, and the current-controlled switch has instead ION and IOFF parameters to denote what current values should turn the switch on or off.  This is described in the Xyce reference guide section 2.3.21.

So instead of putting all of your parameters into the W device's dialog box (W, W_Line_2, etc.), put them into a ".MODEL Section" element with the contents ".model myswitch csw (It=0.01 Ih=0 RON=100Ohm ROFF=10MOhm)" and then change your W device specification in Qucs-s to have a first line that contains the name of a voltage source whose current should be used to determine the switch state, and a second line with "myswitch" on it.  (You could stuff 'em both on the same line, I suppose).  Qucs-s will then generate a legal ngspice line for the switch.

If you want to run it in Xyce, you'll need to modify the switch model card to be of type "iswitch" and use a set of legal parameters for Xyce's switch (which will be different than ngspice's).

tvrusso, 2017-09-27:
> Also SPICE notation should be used for SPICE-only devices. For example SPICE doesn't support measurement units (like Ohm)

Actually, standard SPICE3F5 ignores everything after a number that it doesn't recognize as a multiplier.  So specifying a resistor as "1MEGOhm", "1MEGFarad", "1MEGHenry" or "1MEGBogus" are all the same as specifying it as "1MEG".  Xyce also follows this rule.  I assume, since ngspice is based on SPICE3F5, that it probably also follows this behavior.

It is still safer just to leave off units, so they can't be misinterpreted.  For example, if you specify a huge capacitor as a "1Farad" cap, it'll actually be interpreted by SPICE as a 1e-15 because it'll see the "f" and think "femto."  

This same issue is why Xyce doesn't support "a" as an "atto" multiplier --- many legacy netlists and model cards have units of Amp specified in current sources and device parameters on the assumption that SPICE would just ignore it, and so we can't let "a" mean "1e-18" or all of those old netlists will be unrunnable.

The 1M vs. 1MEG thing is really important, though --- because SPICE netlist parsing is case-insensitive, "1M" and "1m" both mean 1e-3.

closed 2017-09-29 by ra3xdh