ra3xdh/qucs_s #4: SpiceLibComp saves full path to user libraries, making it difficult to share schematics

ra3xdh, 2017-10-01:
_From @tvrusso on April 25, 2017 17:25_

If a user creates a set of User Libraries by putting them into ~/.qucs/user_lib and then uses them in schematics, Qucs-s saves the full path to the library file in the metadata for the part.

If one bundles up one's schematic and libraries and shares them with someone else, they don't display properly, even if that user has dropped the library files into the correct ~/.qucs/user_lib directory.  That's because the full path is no longer pointing at "~/.qucs/user_lib," but rather to "/home/<firstuser>/.qucs/user_lib." This can also bite a user who is running qucs-s on multiple machines with different home directory paths (/users/<myname> on one, /home/<myname> on another, for example).  In fact, that's how I realized this was a problem.  It's easy enough to fix by double clicking on the default symbol that shows up and fixing the path, or by hand-editing the .sch file, but for a large schematic with lots of user-library parts, it could become a problem.

It would be nice if qucs-s could recognize "~" as short-hand for "my home directory" and save this in the library path name in SpiceLibComp parts rather than the actual full path to that specific user's home directory on that specific machine.

_Copied from original issue: ra3xdh/qucs#23_

ra3xdh, 2017-10-01:
Usual `LibComp` in mainline Qucs has the same behavior. If you paste `LibComp` from the user library, it will store the full path too. And the relative path is used for system libraries. It's not so easy to fix as it seems. Replacing the home path by `~` or `$HOME` will solve the problem only for Unix, but Windows will require `%USERPROFILE%`. The proper solution will be to introduce default library search paths and library priorities. It will allow to specify relative library path or even only library name. I will think how to proper fix this limitation.

tomhajjar, 2023-08-31:
It's always best to include any custom/user libraries in the zip file you want to share. Always best to place Spice models, S-parameter and data files in the same project folder.

I've been burned many times trying to simulate schematics without user libraries or S-parameter files.

closed 2024-02-16 by ra3xdh