ra3xdh/qucs_s #7: Snap or appimage Support

BearzRobotics, 2018-01-01:
I was wondering if you ever though about creating a snap or appimage of the project to make it easier to use on other distros.

ra3xdh, 2018-12-14:
Yes, we are currently considering adding an AppImage. If you could help with adding AppImage in OBS, you are welcome. @eldarkg has created a SNAP package for Qucs-S once, you can request him to actualize it.

ra3xdh, 2020-01-20:
I have just added AppImage for 0.0.22 release https://github.com/ra3xdh/qucs_s/releases/download/0.0.22/Qucs-S-0.0.22_x86_64.AppImage It was tested on OpenSUSE live DVD where Qt4 or Qucs were never installed. Simulation using system Ngspice works without problems. This AppImage doesn't include Ngspice!

closed 2020-01-20 by ra3xdh

drtrigon, 2020-04-16:
The AppImage is brilliant, just "saved my live" as Ubuntu 20.04 seems to drop Qt4 support. Thanks a lot!