ra3xdh/qucs_s #8: bug  in linear sweep parameters

btorrem, 2018-01-06:
Hi,
there's a small bug in the linear sweep parameter conversion.
For example a sweep defined from -10 to 15 with a step of 1, 26 points gives the following in 
spice4qucs.dc.cir : .dc v2 -10 15 0.961538, but should be .dc v2 -10 15 1

I'm running Qucs-S 0.0.20 with Xyce 6.8 and trying the diode clipper circuit from the Xyce user guide

the files are attached

[bug_qucs-s_linear_sweep.tar.gz](https://github.com/ra3xdh/qucs_s/files/1608717/bug_qucs-s_linear_sweep.tar.gz)

thanks,
Bruno

btorrem, 2018-01-06:
Hi,
there's a similar bug in the log sweep:
log sweep from 1 to 10, 10 points per decade gives :
 .dc v2 1 10 0.818182 
but should be
 .dc dec v2 1 10 10

Bruno

tvrusso, 2018-01-07:
That's because at each point in qucs_s/qucs/components/param_sweep.cpp, when generating the netlist the "Step" you specify in the param_sweep dialog is ignored, and "step" is computed as "(stop-start)/points" --- which in the case of -10 to 15 and 26 points, gives a step size of 0.961538.  

If you actually want the step to be 1.0 at this stage, you have to make (stop-start)/points be 1.0, which means you want 25 steps.

If you type "25" into the points box of the param sweep, you'll get the behavior you expect at netlist generation time --- but Qucs-s will display the WRONG "step" in the parameter sweep dialog.

I think what's going on is that the dialog box (in qucs/components/componentdialog.cpp) is in general computing points from step (if you enter a step) and step from points (if you enter points) differently, but only start,stop and points are what are actually used by the param_sweep.cpp code when it generates a netlist.  

Down in componentdialog.cpp's "slotNumberChanged" it is clear that "step" is computed as (stop-start)/(number-1) (where number is the number of points given in the dialog), not (stop-start)/number.  But in "slotStepChanged", if you enter a step, "number" is computed as floor((stop-start)/step+1.0).

Because the dialog computes step and number from each other in a different way that the parameter sweep object computes step from number, this can be very confusing.   The two parts of the code (dialog and netlist generation) should be using the same formula for computing step from the other parameters.

tvrusso, 2018-01-07:
For linear sweeps, I think the fix for this is just to change "step = (stop-start)/points" to "step = (stop-start)/(points-1)" in the two places where it appears in qucs_s/qucs/components/param_sweep.cpp.

The way that "Step" is computed from "Start," Stop," and "Number" in qucs_s/qucs/components/componentdialog.cpp is the correct way --- if "Number" is the number of simulation steps, then there are "Number - 1" intervals of length "(Stop - Start)/(Number -1)".  The way it's done in param_sweep.cpp for netlist generation isn't right, or rather, is not generating the correct step size to produce the desired number of simulation steps.

I haven't looked into what the right fix to logarithmic sweeps is.  That bug is not as simple, since it's not correctly generating the ".DC" line to produce a decade sweep, and is creating instead a linear sweep with a bizarre step.