ra3xdh/qucs_s #9: Warning on simulations

sirkha, 2018-01-09:
When running a simulation (ngspice) on the [this schematic](https://github.com/ra3xdh/qucs_s/files/1616279/fault_supply.sch.txt), the warning 
`[Warning!] <prefix>/share/qucs/xspice_cmlib/include/ngspice_mathfunc.inc file not found!` 
is thrown. The file in question exists at 
`-<prefix>/share/qucs-s/xspice_cmlib/include/ngspice_mathfunc.inc`. 

Quickly looking through the code, I couldn't sort out all the QUCS_CORE and QUCSSHARE references, and am not entirely sure what to change to fix this.

tvrusso, 2018-01-09:
@ra3xdh fixed this on the master branch in commit 35a21c71b on Nov 1.    That is the only commit on the master branch *after* the release of qucs_s 0.0.20, so my guess is that you're running 0.0.20.

If you build the current contents of the master branch from source, this warning message will go away.

sirkha, 2018-01-09:
I am running 0.0.20, I was probably looking at the master branch code, too. Thanks.

closed 2018-01-09 by sirkha