ra3xdh/qucs_s #11: Windows 0.0.20 with ngSPICE 27_QucsS error running simulation

computerchemist, 2018-02-06:
Starting a simulation with ngSPICE selected as default simulator, I get the following preamble and error in the simulator window popup:
--------------------------------------------------------------------------------------------
unrecognized parameter (m) - ignored
Warning: Model issue on line 0 :
  .model xop1.xx1:xx1:jmod_j2 pjf(is=1.5e-11 beta=0.0002701 vt0=-1 n=1 lam ...
unrecognized parameter (n) - ignored
unrecognized parameter (m) - ignored
Note: can't find init file.

exit: no such command available in ngspice
Note: No ".plot", ".print", or ".fourier" lines; no simulations run
------------------------------------------------------------------------------------------

I got a similar error using  ngSPICE 26_QucsS (which is still the recommended version for 0.0.20 on the main web page).

The same schematic runs flawlessly in Qucsator and somewhat flawlessly in Xyce (somewhat meaning DC offsets are above Vcc on output for some inexplicable reason in Xyce).

Attached, the project and screenshot
[image]

[mixer test1.zip](https://github.com/ra3xdh/qucs_s/files/1698797/mixer.test1.zip)

tvrusso, 2018-02-06:
Re: unexpected results from Xyce:

I think there's an issue with using that TL071 OpAmp model in ngspice and Xyce.  The OpAmp subcircuit uses two jfets, but with model parameters that are not supported in the other simulators.

The SPICE3F5 JFET model (from which both ngspice and Xyce are derived) have no N or M model parameters, so both complain about this (ngspice has added an M _instance_ parameter as a parallel multiplicity parameter that neither SPICE3F5 nor Xyce have, but it isn't the same as Qucs' gate PN junction grading model parameter, and since that addition is not a model parameter the one given here is still not recognized).  

The fact that these parameters are ignored is probably why Xyce's results differ from those of Qucsator --- the qucsator JFET model uses that parameter in its gate capacitance model.  Fact is, this makes the SPICE and Qucsator JFET models not equivalent, so the TL071 opamp model in qucsator isn't likely to do the same thing in either Xyce or ngspice.

Re: ngspice refusing to run your circuit:
I just tried running your schematic through ngspice 26 on my Unix box, and it worked just fine, producing similar results as Xyce.  This may be an issue with the specific Windows build of ngspice you're running not dealing properly with the commands in the .control section.

Here's what my ngspice run did (first few lines, showing that it actually got to transient, unlike yours):
The simulate window:
[image]

Xyce and ngspice data plotted next to each other showing similarity of behavior, which is unsurprising because they both implement (nearly) the same JFET model:
[image]

The netlist produced by qucs-s has this in the control section:

    .control
    echo "" > spice4qucs.cir.noise
    echo "" > spice4qucs.cir.pz
    tran 2.5e-05 0.005 0  
    write mixer_test1_tran.txt v(in) v(out) v(raw) 
    destroy all
    reset
    
    exit
    .endc

Clearly the recommended ngspice for Windows isn't handling that section the same way that ngspice on a unix box is handling it.  It might not be entirely qucs-s's fault.

ra3xdh, 2018-02-06:
Windows version of Ngspice should be installed strictly at the `C:\Spice` directory. Otherwise Ngspice will not find XSPICE models and `spiceinit` file. For example, alias for `quit` and `exit` is defined here. The `.control` section of the netlist  is produced correctly. Your circuit runs correctly on my Windows system and produces the same result as on Linux. Please check your installation of Ngspice.

closed 2018-02-06 by ra3xdh

tvrusso, 2018-02-06:
Just a quick additional note:

I tried dropping TI's PSpice model of the TL071 into your schematic (available here:  http://www.ti.com/product/TL071/toolssoftware), modifying the syntax so it was actually acceptable to Xyce (attached).   (I put this file into .qucs/user_lib as TL071.lib so it appears as a component in my Libraries tab, and when I placed it in the circuit I selected opamp5t as the SymPattern).  Gotta watch the placement --- the opamp5t symbol has the + and - inputs inverted from the way it's done in the TL071 part, so it takes some additional wires to get this hooked up the same way you had it.

[TL071.txt](https://github.com/ra3xdh/qucs_s/files/1700229/TL071.txt)

Unfortunately this model uses the POLY form of F and E sources, which is apparently only available in ngspice with XSPICE extensions.  Whether it will work for you depends on how your ngspice was built.  Xyce does support that format.  It worked in my ngspice build, and produced the same results as Xyce.

[image]

What I got was very, very different behavior than when I ran it with the Qucs TL071 model..  I do not have qucsator installed on my system, so don't know if this is any better.
[image]

computerchemist, 2018-02-07:
Wow, thanks for all this feedback!
ra3xdh - I didn't spot the fact that the special ngspice build needed to be in c:\spice. I must have missed that vital info on the docs somewhere, thanks.
tvrusso - Thanks for confirming my strange results. I would have expected the outputs to have been in line with Qucs (Qucsator gave me the same as Qucs) - indeed, a physcial breadboard game me similar results. 
I should have split my bug in two - the strange results with Xyce and ngspice (blue Vin is above Vcc - it's uncoupled from the DC offset by C1 yet is clearly offset -- and red output for Vout is flatlining) you seem to have nailed down to the TL071 model, so it's clearly not qucs_s's fault. I came looking for Qucs_s as I was having lots of hanging issues with standard Qucs when Vin went above 1V. Your last result is absolutely what I would have expected.

tvrusso, 2018-02-07:
Delighted to help, and always happy to see someone trying to use Xyce with qucs-s.

If you use Xyce a lot, you might consider joining the xyce-users group on Google Groups.  The Xyce developers hang out there and are keen to answer community questions.

When qucs-s runs Qucsator, it is using the Qucsator installed by qucs --- which would explain why qucs-s+Qucsator gives you the same result as qucs (which ONLY can run Qucsator).

It does point out, though, that some of the parts models in the Qucs library aren't correct to use with ngspice or Xyce, especially if they make use of Qucs-specific models under the hood --- so one ignores warnings about unknown parameters at one's peril.  That's why my first thought was to look for SPICE models of the TL071 from other sources right away.

tvrusso, 2018-02-07:
The modified TL071 file was uploaded here and linked to in my earlier comment, called "TL071.txt".  

This file is the same model that you can download from the TI link I provided in that same comment, with minor syntax fixes (Xyce treats all lines that begin with a space as comments, as did SPICE3F5, and the POLY E source in Xyce does not accept node pairs as (a,b)).

To use that modified file, just put it in a folder called "user_lib" inside the ".qucs" folder that qucs-s creates, and rename it to "TL071.lib" so you can drag-and-drop it into schematics from the "Libraries" tab.

computerchemist, 2018-02-07:
Thank you for your time explaining - it really wasn't obvious from an eyeball, but I'm a fast learner :)
I can confirm your amended TI 071 model with the modified ngspice in the right directory on windows(!) now gives me the expected results, the same as qucsator but much much faster.