ra3xdh/qucs_s #13: Some models from qucs-s library not getting "normalized" parameters, either generate unparsable netlist line, or lose scale value

tvrusso, 2018-02-19:
I just attempted to use some of the LED models in the Qucs library in a Qucs-s schematic with Xyce.  These models all have spaces in the diode parameters between value and units, and a 50 MOhm leakage resistor.  The specific LED model I tried to use was the green one.

Qucs-s is generating a netlist that doesn't squeeze out the space between parameter and units of the diode parameters, resulting in a parse error from Xyce (which uses space as a field separator, like the original SPICE does).  Further, the "50 MOhm" is not converted to "50 Meg" as it seems it should be by the use of "spicecompat::normalize_value" in the resistor.cpp code.

The attached schematic and the netlist it produces are in this tar file:
[led_prob.tar.gz](https://github.com/ra3xdh/qucs_s/files/1737658/led_prob.tar.gz)

A quick glance at code makes it appear that when the parameter list for the diode is generated, nothing is calling "normalize_value".  If I modify (by hand) the diode model in LEDs.lib to take out the spaces, that fixes the netlist so it runs.  But the "50 MOhm" resistor becomes a 50 ohm resistor.  

If I remove the space from the leakage resistor specifier in LEDs.lib, qucs-s emits "50MOhm" and not "50Meg".

But qucs/components/resistor.cpp *does* call spicecompat::normalize_value in its spice_netlist function, and still doesn't convert "50 MOhm" into "50Meg" as I would expect it to.  I don't quite understand what's going wrong there.

ra3xdh, 2018-02-19:
If the model contains subcircuit in original Qucs libraries, it is represented as Qucs netlist, but not as XML subcircuit. The `spicecompat::normalize_value` is invoked only for "normal" components. If Qucs-S finds Qucs netlist, it redirects it to `qucs2spice` processor (located at `extsimkernels/qucs2spice.cpp`) that will convert it to SPICE netlist. It is a simple Regexp-based netlist converter and some functions could be missing here.  Probably `qucs2spice` would be benefit from redesign. So the `Resistor::spice_netlist` is not invoked for such library component.

tvrusso, 2018-02-19:
Ah.  Thanks for clarifying where I should look.  I'll look at qucs2spice.cpp soon.

tomhajjar, 2022-08-15:
I looked into the LED library issues. I don't understand why the library even uses Ohm, v, uA, pF...etc. I deleted all "units" and converted say 1Meg to 1e6, 1uA to 1e-6...etc. 

I also deleted all parameters that caused warnings in ngspice. Didn't check xyce yet. Almost every one of these parameters had values of "0" anyway. There are numerous parameters that have values of "0" but didn't generate warmings so I ignored them.

I don't understand why a generic LED even needs such complex models to begin with. For sure they are wrong for any LED I might buy anyway. There are literally 1000s of "RED" LEDs with different Spice parameters. If we actually need detailed data for specific LEDs we would use the manufacturers provided Spice model. The only LED I would actually need detailed data for is a high power white LED for lighting or a LED used for say fiber optics. 

[LED_prj.zip](https://github.com/ra3xdh/qucs_s/files/9334493/LED_prj.zip)

ra3xdh, 2023-06-02:
LEDs library has been fixed by #134. I have also provided some other new libraries that have better compatibility with SPICE. Qucsator netlist syntax should be removed from libraries and SPICE entries provided instead. I am considering the update of ideal devices libraries #258 This issue could be closed.

closed 2023-06-02 by ra3xdh