ra3xdh/qucs_s #14: Integrating qucsator

aishaque, 2018-09-16:
Hi,

My OS is Ubuntu 18.04 running on HP Zbook 17 G#.  I have installed qucs_s 0.0.20 using the .deb file for Ubuntu 16.04. I am able to run ngspice, Xyce and opusspice simulations. In order to run qucsator I gave the executable path for qucsator installed by version 0.0.19 of qucs-spice, which is available in Ubuntu software. When I run the simulation I get the following error: 

/snap/qucs-spice/4/usr/bin/qucsator: error while loading shared libraries: libqucs.so: cannot open shared object file: No such file or directory. This lib file is installed in /snap/qucs-spice/4/usr/lib/.

How can I integrate qucsator simulator in qucs_s?

Amir

ra3xdh, 2018-09-16:
Hello Amir, snap probably uses relative paths to libraries. When you are trying to run qucsator outside of snap, it cannot load libraries. You need either to create some custom script that will add the `/snap/qucs-spice/4/usr/lib/` path to `LD_LIBRARY` path variable, either to use DEB package for basic Qucs. DEB installation installs libraries in system `/usr/lib64` location.  There was an Ubuntu repository for main Qucs project somewhere at launchpad.net

aishaque, 2018-09-21:
I removed the snap Qucs and from the GitHub repository. This way I can access the qucsator from qucs_s by giving the path. 

Interestingly, Qucs built from git repository does not give the option to add 3rd party simulators like ngspice. Since I am using qucs_s, which does I guess I don't need to fret over this.

closed 2018-09-21 by aishaque