ra3xdh/qucs_s #16: Spice Opus not loading external libraries

alexpacini, 2018-11-21:
When I simulate with Spice OPUS using the spice netlist file component I always get an error:
```
Error: Failed to open "/home/alex/.qucs/1_prj/c3d1p7060q.lib" in file spice4qucs.cir on line 33
```
I tried to remove the quotes and the simulator is not complaining anymore.
Does this works out of the box for you?
You can try with any spice model.

A side question: how to get a steady state response simulation with qucs?

ra3xdh, 2018-12-14:
I can confirm this. SpiceOpus cannot process the quotes in `.INCLUDE` statement. I can enable automatic processing of this in future versions.

The `ssse` analysis currently could be supported only using the "Nutmeg script" analysis type. Check our presentation from the MOS-AK Workshop (Slide 14) for more details http://www.mos-ak.org/dresden_2016/presentations/T2_Brinson_MOS-AK_Dresden_2016.pdf Unfortunately, we have no better documentation for this analysis type now.

mhx-gh, 2020-06-20:
This is off-topic, but ssse sounded interesting. The quoted MOS-AK.. presentation has nothing pertinent on slide 14 (output spectrum of a non-linear transformer + synchronous rectifier).

ra3xdh, 2022-02-19:
Fixed at the `qt5` branch by https://github.com/ra3xdh/qucs_s/commit/dae16575981202268b004fafe913ba832b120efd

closed 2022-02-19 by ra3xdh