ra3xdh/qucs_s #20: ngspice op amp model clipping not working correctly

cactorium, 2018-12-17:
Hello! It looks like the op amp clipping isn't working correctly during saturation, at least using the ngspice backend; it seems to do this weird thing where the output gets stuck to the voltage rails instead of returning to normal operation, and the op amp seems to saturate in the wrong direction somehow; when the clipping starts it jumps to the positive rail instead of saturating on the negative one. Oddly enough, the clipping seems to work correctly if I flip the positive and negative inputs, and the op amp works when it's not clipping with the inputs plugged in either way. Maybe the netlist generator has the two inputs mixed up or something? I've attached a simple test circuit below:

[test.zip](https://github.com/ra3xdh/qucs_s/files/2688208/test.zip)

ra3xdh, 2019-06-23:
Yes, this is a bug in netlister. Waiting for #27 for fix.

ra3xdh, 2019-06-23:
Fixed by #27

PR #27 (2019-06-23): Correct input pins of the opamp component by swapping them.
Fixes #20.
closed 2019-06-23 by ra3xdh