ra3xdh/qucs_s #21: Expired Keys in the build system

lrenaud, 2019-02-03:
After following the instructions and installing the keys to use the package repositories as outlined in [the instillation instructions](https://ra3xdh.github.io/#inst) requesting an update for the package lists on Ubuntu 18.04 yields the following warning.

It looks like the Release.key being uploaded has expired.

`W: GPG error: http://download.opensuse.org/repositories/home:/ra3xdh/xUbuntu_18.04 ./ Release: The following signatures were invalid: EXPKEYSIG 184B98BE3E68CD11 home:ra3xdh OBS Project <home:ra3xdh@build.opensuse.org>
E: The repository 'http://download.opensuse.org/repositories/home:/ra3xdh/xUbuntu_18.04 ./ Release' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
`

ra3xdh, 2019-06-23:
Sorry for delayed answer, I have no Ubuntu system now to reproduce this bug, but the Debian repository works correct. I have found an information that `Relase.key` files are generated automatically by OBS and this issue coulbe self-fixed. Otherwise use manual installation from the DEB binary package.

closed 2019-06-23 by ra3xdh

qaxi, 2019-07-08:
Still the same error.

The key is not generating automatically, because it is your personal key.

```
pub   rsa2048 2016-11-08 [SC] [expired: 2019-01-17]
      69B1 C091 6FDB 15A8 7417  43E8 184B 98BE 3E68 CD11
uid           [ expired] home:ra3xdh OBS Project <home:ra3xdh@build.opensuse.org>
```

Please see https://forums.opensuse.org/showthread.php/518046-How-to-update-expired-OBS-home-project-signing-key

pebbe, 2021-03-20:
I get the same error. Please reopen this issue.