ra3xdh/qucs_s #22: No digital components

jdav63, 2019-03-19:
I'm using qucs-s 0.0.21 on Linux-64. I followed the instructions to install from the debian repo and successfully installed qucs-s. However when following the getting started guide section on digital simulation I'm unable to find an entry in the component combobox for digital components. I've installed freehdl (though not sure whether that should make any difference). Don't know if this is a bug or what, but any advice appreciated.

ra3xdh, 2019-03-19:
Digital simulation with FreeHDL is not implemented in Qucs-S. Special mixed signal components could be added in future, but not in the next release.

closed 2019-03-19 by ra3xdh

jdav63, 2019-03-20:
Thanks for the feedback. Then shouldn't this at least be regarded as a documentation bug? Currently quc-s links to a document which tells you how to use a feature which isn't actually implemented...