ra3xdh/qucs_s #28: Xyce script HB data post processing

kmihaylov, 2019-06-29:
Hello!

I'm trying to use Qucs-S with Xyce to do a HB simulation of an impedance of a diode. Attached is my schematic.
I'm following the manual, chaprer "Xyce simulation output data post processing with the Xyce script component and SPICE .PRINT statements".
I can do the simulation (I found the HB output in C:\Users\%user%\.qucs\FILENAME.txt.HB.FD.txt)
However I can not plot anything from it after ismulation finishes.
Otherwise if I use AC simulation I can get the Xyce result in Qucs-s and to prepare some graphs.
Maybe I'm doing something wrong? Please point me how can I use che 'Xyce script' output directly.

Thank you!

[image]

tvrusso, 2019-06-29:
As a Xyce developer I would love to be able to give you a definitive answer, but these days I am unable to build qucs_s on any of my systems because all of those systems have dropped support for Qt4, and qucs/qucs_s are unable to be built with Qt5.  So I am unable to try out anything I suggest, and have to go only on what I can read in the qucs_s source code.

However, I've looked over the code that handles reading HB output, and it is explicitly looking for the "HB.FD.prn" suffix that Xyce would have used on the file name itself.  This is probably why you can't get it to display the output, because you're explicitly naming your output file name "hb.txt" (which Xyce tweaks into "HB.FD.txt" and "HB.TD.txt" for its frequency- and time-domain outputs).

You can see this code in the qucs/extsimkernels/abstractspicekernel.cpp file if you search for "HB.FD", where you'll find:
```
        if (ngspice_output_filename.endsWith("HB.FD.prn")) {
            parseHBOutput(full_outfile,sim_points,var_list,hasParSweep);
            isComplex = true;
            if (hasParSweep) {
                QString res_file = QDir::convertSeparators(workdir + QDir::separator()
                                                        + "spice4qucs.hb.cir.res");
                parseResFile(res_file,swp_var,swp_var_val);
            }
        }
```

Can you try changing your Xyce script to use "hb.prn" for the output file name, and see if that then enables you to access the data in qucs diagrams?  That may be enough to get your simulation results plotted through Qucs_s's plotting features.

tvrusso, 2019-06-29:
Looking over the Xyce Script code as well, I see that there is a property for the object called "Outputs" that allows you to specify what output files the script will create, and that qucs_s must be aware of.

If you change your print statement to output to a file with a ".prn" suffix *AND* add the expected HB output file to that "Extra outputs to parse" property it may work.  If you give "hb.prn" on the .print line, Xyce should be producing "hb.HB.FD.prn" and "hb.HB.TD.prn" output files.  You should add the first name to that "Extra outputs to parse" property in order for qucs_s to know to read it.  And the output file has to have that "HB.FD.prn" suffix in order for qucs_s to realize it's HB output and know how to parse it.

tvrusso, 2019-06-29:
I just installed qucs_s from a binary installer and Xyce on a Windows machine and tried out my own suggestions, in the hope that they might actually have worked.  They did not.  Worse, I've gotten it to crash repeatedly when trying to run this schematic with the HB.FD.prn file correctly specified.

Since I can no longer actually work on qucs_s source myself, I'll have to leave this to Vadim or Maria to comment further on what is going on here.

To make it easy, I've created a .sch file with what I think should work according to my reading of the code, and maybe that can help debug? 

[Issue28.zip](https://github.com/ra3xdh/qucs_s/files/3342204/Issue28.zip)

tvrusso, 2019-06-29:
I've spent a little more time looking at this and figured out what the issue is that leads to the crashes on the schematic I just posted.  It's that qucs_s is unable to handle the use of expressions in the .print hb line.  

The code that tries to read in the HB output and convert it into a qucs data set  is assuming that all variables output in the output file's header are of the form Re(variable) or Im(variable), and trying to strip off the Re() and Im() part to get the raw variable name.  It does this trimming simply by stripping off the first three characters and dropping the final character of the header field, irrespective of whether it is *actually* "Re(variable)" or not.  Re(variable) and Im(variable) are only emitted when you try to output the variable directly (e.g., by ".print hb V(1)", which will result in two columns, Re(V(1)) and Im(V(1)).

That means is completely confused by the brace-delimited expression syntax, which never has that Re() and Im() pattern.  So when you run the Issue28.sch circuit I attached, which properly identifies to qucs_s the actual name of the HB output file it should examine, it crashes.  

This is due to limitations of the parsing done in the function parseHBOutput in file abstractspicekernel.cpp in qucs/extsimkernels.

So I had been on the right track with why qucs_s was producing empty datasets: it is indeed necessary to make sure that the Xyce Script object has the right list of filenames to scan, but getting that right exposed a real bug in how the HB output is parsed.

However, if you change the .print so it just prints v(input), v(output), and I(VPr1), you'll get the correct output on the .HB.FD.prn file that qucs_s can then process and plot.  But you won't be able to plot the quantities you want (the expressions like vr(input)/im(vpr1)) unless there's a way to make qucs_s do that computation itself (I'm not familiar enough with qucs plotting capabilities to know if that feature exists).

You could always import the .HB.FD.prn file into another tool and plot it there, if you make sure NOT to tell qucs_s to parse the file (that is, DON'T give its correct name it in the Outputs property of the Xyce Script object, otherwise qucs_s will crash).  This attached netlist will run and allow you to plot the real part of the three variables, but qucs_s does not appear to have any way of allowing the plot of the imaginary parts that I can see.  Perhaps Vadim or Maria can shed some light on that.

[Issue28_noexpr.zip](https://github.com/ra3xdh/qucs_s/files/3342337/Issue28_noexpr.zip)

kmihaylov, 2019-06-29:
I am following your posts all day. Thank you for your efforts and spent time!

tvrusso, 2019-06-30:
Sorry to say, though, that at this point I can no longer help other than to suggest using a tool other than qucs_s to plot the data from Xyce until someone has a chance to improve the parseHBOutput function and any downstream functions that may also be having trouble with the variable names that result from its action (where the actual crash may be happening --- I can't tell, and can't build qucs_s for myself anymore to debug it further).

Personally, I like gnuplot for the heavy lifting plot work, but just about any program that can read columnar data should work.  You could also request CSV output from Xyce (FORMAT=CSV on the print line) and import it into Excel if you were really hurting.

ra3xdh, 2019-06-30:
I had a quick look at this. Everything works correct with Xyce-6.6.0 on Linux platform using the filename `hb.prn` (auto-recognized). Newer Xyce version cannot be installed on my Slackware-14.1 host, and currently I cannot to reproduce this issue. As temporary solution I may only suggest to switch to old Xyce version. Also why not to use an usual HB simulation? It could be found in Simulation->Harmonic Balance.

kmihaylov, 2019-06-30:
The fact that it produces the output data is enough for me. Graphical representation helps me work faster.

Normal HB simulation works w/o post-processing, and I would like to calculate the input impedance of the network (U/I). Therefore I used Xyce script. Usually I use normal equations, but with Xyce there are some limitations as I understoot?

Excuse me for the off-topic, but I have some concerns about the simulation shown in the 1st post. I'm always getting similar input impedance, despite I use 0.05 V and 10 V as a signal amplitude. Given that the diode is non-linear element, isn't it expected for the input current to express non-linear change related to the input voltage (below and above the junction voltage)?

I have no understanding of small-signal and large-signal simulations. From the documentation the HB is described as large signal sinulation. Should the small-signal AC analysis give different result?

Thank you very much!

tvrusso, 2019-06-30:
I just constructed a netlist from your schematic manually and ran it using a .step over the input voltage  from 0.5 volts up to 10 V in steps of 0.5 to produce a family of solutions.  This netlist should be close to what qucs_s would have generated from the schematic.

I clearly see the impedance values at 2GHz changing as the source voltage changes as one would expect.

I'm attaching a zipped version of that netlist that you can run in Xyce directly (i.e., from a command line prompt without qucs_s).  It uses the "gnuplot" format output, which is the same as the .prn format with extra white space between the various blocks of values representing a stepped value.  You could change it up a little to plot some other way.

AC analysis is indeed different, in that it analyses behavior of the circuit equations that have been linearized around the DC operating point.  HB is solving the full nonlinear problem in the frequency domain.  They would really only show the same results if the signal is small enough to make the linearized problem a good approximation to the nonlinear.

Since this has strayed from the qucs_s issue, though, perhaps it might be better continued on the Xyce google group (https://groups.google.com/forum/#!forum/xyce-users).

I am also concerned by Vadim's statement that he can't run Xyce after 6.6 on Slack.  I would be happy to help with that, too.  Xyce should be portable to all linux distros (although satisfying its dependencies can be challenging).  6.6 is at this point quite old and missing a lot of features.  If you'd like a hand with that, please post to the Xyce google group

I can only run qucs_s these days on a windows virtual machine, which is where I was seeing those crashes, and only when using the expressions on the print line.  Perhaps the crashes don't happen on Linux at all and it Just Works.  But reading through the source code for the parseHBOutput function doesn't make it look like it's getting variable names correct out of the HB header unless they're simple V(node) or I(device) outputs, because of the way it tries to strip off the "Re()" and "Im()" text (which aren't there for expressions).
[diodeimpedance.zip](https://github.com/ra3xdh/qucs_s/files/3343093/diodeimpedance.zip)

ra3xdh, 2019-06-30:
> I am also concerned by Vadim's statement that he can't run Xyce after 6.6 on Slack

Slackware-14.1 is outdated and unsupported. I cannot run precompiled binaries, probably it could be compiled from sources. The Xyce should be run correctly on the latest Slackware-14.2, but I didn't test it.

ra3xdh, 2022-02-19:
I can reproduce the crash with `issue28.zip` now with Xyce7.4 and the latest Qt5 build.

ra3xdh, 2022-02-19:
The problem is that the parser assumes that all variables in the HB output are complex and have Re and Im parts. Real variables in the HB output are unsupported now.

ra3xdh, 2022-03-20:
The XYCE HB dataset may contain not only complex but also a real varibales. The solution may be to use the common STD dataset parser instead of the existing custom HB dataset parser in `AbstratSpiceKernel` class. I have just added a fix on the `current` branch. Now the provided test circuit could be simulated without a crash.

ra3xdh, 2022-03-20:
The issue has been fixed on the `current` branch. Closing this.

closed 2022-03-20 by ra3xdh

kmihaylov, 2022-03-21:
Thank you Vadim!
73 de LZ1MZK