ra3xdh/qucs_s #31: initialDC property default value

leon-vv, 2020-01-11:
When doing a transient analysis the _initialDC_ property is set to _yes_ by default. This leads to a very confusing user interface for a user that's not aware of this option. Initial voltages and currents for capacitors and inductors can still be set but are by default ignored. I propose therefore to set this option to _no_ by default.

Qucs seems to completely ignore this option, and always performs the simulations with the initial conditions supplied by the user.

Also, thanks for all the work put into this software, it's a big improvement of Qucs. Are there any plans to upstream this code?

ra3xdh, 2020-01-19:
The initial DC=yes is default for SPICE simulators. So this default setting should remain unchanged. It's required for 90% of semiconductor circuits. Otherwise the simulation may be broken. 

Concerning the merging plans, the https://github.com/Qucs/qucs/pull/275 was refused. We decided to port the SPICE support subsystem to new modular architecture. But more than two years passed and this architecture still be not ready. So, the probability to see this stuff merged in the near future is very low.

closed 2020-01-19 by ra3xdh

leon-vv, 2020-01-20:
> The initial DC=yes is default for SPICE simulators. So this default setting should remain unchanged. It's required for 90% of semiconductor circuits. Otherwise the simulation may be broken.

Okay, that's fair. Still I feel like there should be a way around the confusing interface. Maybe disable the initial capacitor and inductor values input if `initialDC=yes` with a message about this property?