ra3xdh/qucs_s #32: Qt5

sizeofvoid, 2020-02-01:
There are any plans to switch from qt4 to qt5?

zcx2020, 2020-05-16:
Have you tried to compile and run qucs-s-0.0.22 with Qt5.x?

ra3xdh, 2020-05-16:
@zcx2020  Qucs-S will not compile using Qt5. Qucs-S shares a Qucs code, which depends on Qt3Support classes. These Qt3Support was dropped in Qt5, and compilation of such projects fails on Qt5. So, currently it is not clear how to port Qucs and Qucs-S to Qt5.

zcx2020, 2020-05-16:
Yes - I just tried it ;)

$ cmake .. -DCMAKE_PREIX_PATH="/opt/Qt5.13.0/5.13.0/gcc_64" -DQt_DIR:PATH=/opt/Qt5.13.0/5.13.0/gcc_64/lib/cmake/Qt5 -DCMAKE_INSTALL_PREFIX=~/qucs-s-0.0.22.qt5
-- The CXX compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring Qucs: VERSION 0.0.22
-- qucs-suite /home/nick/qucs-s-0.0.22.qt5
    /home/nick/qucs-s-0.0.22.qt5/build
-- Looking for Q_WS_X11
-- Looking for Q_WS_X11 - found
-- Looking for Q_WS_WIN
-- Looking for Q_WS_WIN - not found
-- Looking for Q_WS_QWS
-- Looking for Q_WS_QWS - not found
-- Looking for Q_WS_MAC
-- Looking for Q_WS_MAC - not found
-- Found Qt4: /usr/bin/qmake-qt4 (found version "4.8.7") 
-- Configuring qucs-s (GUI): VERSION 0.0.22
-- Found Qt4: /usr/bin/qmake-qt4 (found suitable version "4.8.7", minimum required is "4.6.3") 
-- The C compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Configuring qucsactivefilter (GUI): VERSION 0.0.22
-- Found Qt4: /usr/bin/qmake-qt4 (found version "4.8.7") 
-- Configuring qucsfilter (GUI): VERSION 0.0.22
-- Configuring qucslib (GUI): VERSION 0.0.22
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_PREIX_PATH
    Qt_DIR

-- Build files have been written to: /home/nick/qucs-s-0.0.22.qt5/build

Falls back to Qt4.8.7 then builds OK.

Appears to run using Qt5.13.0...

$ LD_LIBRARY_PATH=/opt/Qt5.13.0/5.13.0/gcc_64:$LD_LIBRARY_PATH ~/qucs-s-0.0.22.qt5/bin/qucs-s

Thanks for the explanation.

cederom, 2021-02-09:
+1 to move to QT5 and maybe even QT6 as its already out. That would allow bring this amazing software back to FreeBSD :-)

Is Quc-S direct dependant on Qucs in a way that when Qucs gets ported to QT5 Qucs-S will also work straight away? :-)

sizeofvoid, 2021-02-09:
> +1 to move to QT5 and maybe even QT6 as its already out. That would allow bring this amazing software back to FreeBSD :-)
... not only FreeBSD. All other Linux Distrubtion would start to package this software.

kmihaylov, 2021-02-20:
+1 for Qt5 since I just tried to build qucs-s on FreeBSD and I got the dependency error. Sure we don't need to force the creator of qucs-s to do this, but my coding abilities are limited.

kwmartin, 2022-02-04:
I have an interest in helping on the port to Qt5. I have never previously helped on an open-source github based development. I have experience with PyQt and have ported some of my Gui's from PyQt4 to PyQt5. I have experience with C, very little with C++.
Question: can someone give me instructions on how best to get started to help out and not "re-invent the wheel" since it looks like a lot of the heavy lifting has already been done? I'm assuming I need to pull a particular github branch and work on a local copy? If so, I probably just need to know which branch? Thanks.

ra3xdh, 2022-02-04:
@kwmartin , thank you for the interest in project! The best way is to unfork Qucs-S and merge it with the main Qucs project. There exists so-called "modular Qucs" https://github.com/Qucs/qucs/tree/modular It already compiles with Qt5, but many things still be missing. The Qucs-S part is intended to be a plugin for basic Qucs. The most probably this approach will be accepted for future releases.

Unfortunatelly Qucs-S is not fully compatible with new modular architecture of the Qucs. Qucs-S was started a long tome before the modular architecture was introduced in the main project. It will be needed to port a code from Qucs-S to Qucs plugins manually and probably rewrite some parts from scratch. It is not possible to simple copy-paste the code responsible for netlisting and result processing from Qucs-S to modular Qucs. If you would like to help you may start from the learning of the "modular" branch in the main Qucs repository. 

I am also considering to take this branch  https://github.com/Qucs/qucs/tree/qt5-19 as the base for Qucs-S porting to Qt5.

kwmartin, 2022-02-05:
Thanks Vadim, this is exactly what I was looking for. I am committed and 
plan on starting around mid week next  week. There will be radio silence 
late January- early February while  i get a sunburn, but if I set my 
laptop up well, I might be able to makesome progress in between catching 
rays. Thanks again, this really helps. Finally, is there any chance you 
tell me the flags cmake requires to compile for debug? (-g -O0 ?) and 
finally, finally, can you point to the file and function where the 
net-lister is entered that I need to set up a gdb breakpoint to start 
stepping my way through the netlister. Thanks again.

ra3xdh, 2022-02-05:
Hello Kenneth, 

> Finally, is there any chance you tell me the flags cmake requires to compile for debug?

you should pass the `-DCMAKE_BUILD_TYPE=Debug` key to cmake to ensure debug-mode. This options will activate the necessary flags for GCC. Also I recommend to open the `CMakeLists.txt` in QtCreator where you can activate Debug mode and set installation prefix etc.

>  can you point to the file and function where the net-lister is entered that I need to set up a gdb breakpoint to start stepping my way through the netlister

The spice-related code is located in the followind subdirectories:
* `qucs/extsimkernels` -- the netlisters and result parsers
* `qucs/spicecomponents/` -- specific devices

The entry point for the netlister is `Ngspice::createNetlist()`. The simulator is started from `Ngspice::slotSimulate()`. Both methods are defined in the `ngspice.cpp` file. The same for Xyce, but defined in `xyce.cpp`. The `ExternSimDialog::slotProcessOutput` is responsible for SPICE raw file (simulation result) processing. Some netlisitng routines common for all SPICE simulators are placed in `AbstractSpiceKernel` class.

ra3xdh, 2022-02-05:
I have recently evaluated the "modular" and latest "non-modular" Qucs branches that could be compiled with Qt5. Now I am more inclined to take the non-modular https://github.com/Qucs/qucs/tree/qt5-19 as the base for Qucs-S porting. This approach will allow to directly copy the SPICE specific code in the source tree and provide some operational application in the near future.

ra3xdh, 2022-02-14:
Hello all,

I have recently started the porting of Qucs-S to Qt5. I have imported the necessary compatibility classes from the Qt3Support module of the Qt4. This is not a good way to implement Qt5 support, but it will allow to compile Qucs-S on modern distributions and build the packages in the near future instead of usage AppImage. The code could be found in the https://github.com/ra3xdh/qucs_s/tree/qt5 branch. The proxy classes are located in the `qucs/qt3_compat` subdirectory. Currently only `qucs-s` application (sources located in `qucs` subdirectory) compiles with Qt5, but it is 90% of work. The basic features of schematic editor, simulation, and vizualization are operational. It remains to port utilities and fix bugs. I hope to finish work within next months and prepare `0.0.23` release. The attached screenshot shows the main window of the Qucs-S application compiled with Qt5.

[image]

ra3xdh, 2022-02-14:
@kwmartin You can use the https://github.com/ra3xdh/qucs_s/tree/qt5 branch as the reference for your work. Currently it could be compiled and run with Qt5. The utilities are not ported to Qt5 yet. Your contribution is welcome.

ra3xdh, 2022-02-14:
See also https://github.com/ra3xdh/qucs_s/issues/72 to track the progress and report issues

ra3xdh, 2022-02-20:
The `master` and `current` branches are switched to Qt5 now. The `qt5` branch is not in use anymore. The legacy Qt4 code could be found in `legacy_qt4` branch. I am closing this issue. Refer to #72 to track the remaining Qt5 migration tasks.

closed 2022-02-20 by ra3xdh

cederom, 2022-02-21:
Quick test confirms `master` build and run fine on FreeBSD :-)
THANK YOU @ra3xdh !! :-)