ra3xdh/qucs_s #33: keeping qucsator and ngspice projects separate

G3VNC, 2020-02-01:
Been using qucs here for about 5 years, mainly to simulate RF networks using S-parameters.

Recently installed qucs-s-0.0.22 (compiled on Linux Mint 19.3) and been getting to grips with ngspice.

As well as new ngspice models, I've accumulated a sizeable archive of models that only run using qucsator.

While it is very convenient to be able to switch between qucsator and ngspice inside qucs-s (albeit with a restart), at startup there's a huge flat list of all the models in ~/.qucs.

Thought about making new projects called QS_<project_name>, NG_<project_name> then copying all the files across manually to sort and separate them, but it seems like a kludge (and a lot of work).

Is there a more elegant way of separating the qucsator and ngspice models so I can pick them from inside qucs-s?

ra3xdh, 2020-02-01:
No, Qucs-S cannot determine if the schematic should be run with Ngspice or Qucsator automatically before the simulation. Currently there is no way to distinguish these circuits automatically.

G3VNC, 2020-02-01:
Thanks.  I'll use the naming convention workaround.

Am I right that there is no way to rename projects, except to create a new one and then copy files across?

G3VNC, 2020-02-01:
Thanks.  Will use the naming convention idea.

73

closed 2020-02-01 by G3VNC