ra3xdh/qucs_s #34: qucs-s-0.0.22 - qucsator crashes using Component library devices

G3VNC, 2020-02-02:
qucs-s-0.0.22 - qucsator crashes using Component library devices

PLATFORM NOTES

qucs-s-0.0.22 on Linux Mint 19.3 x64

$ which qucsator
/usr/local/bin/qucsator
$ /usr/local/bin/qucsator --version
Qucsator 0.0.18
Copyright (C) 2003-2009 Stefan Jahn <stefan@lkcc.org>
Copyright (C) 2006 Helene Parruitte <parruit@enseirb.fr>
Copyright (C) 2006 Bastien Roucaries <roucaries.bastien@gmail.com>

This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BRIEF DESCRIPTION

Models using active devices from Component Library crash qucsator.

STEPS TO REPRODUCE ERROR

Set Default simulator to ngspice.

Restart qucs-s.

Create a simple model using 1N4148 from Component library.

e.g.

# Qucs 0.0.22  ~/.qucs/Tutorial_prj/test_lib_qucsator_diode.sch

R:R1 _net0 _net1 R="10k Ohm" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85"
Diode:D_1N4148_1 gnd _net0 Is="222p" N="1.65" Cj0="4p" M="0.333" Vj="0.7" Fc="0.5" Cp="0" Isr="0" Nr="2" Rs="68.6m" Tt="5.76n" Ikf="0" Kf="0" Af="1" Ffe="1" Bv="75" Ibv="1u" Temp="26.85" Xti="3.0" Eg="1.11" Tbv="0.0" Trs="0.0" Ttt1="0.0" Ttt2="0.0" Tm1="0.0" Tm2="0.0" Tnom="26.85" Area="1.0"
Vac:V1 _net1 gnd U="10 V" f="1 GHz" Phase="0" Theta="0"
.AC:AC1 Type="log" Start="1 kHz" Stop="10 MHz" Points="1000" Noise="no"

F8 - all OK
F2 - all OK.

Change Default simulator to qucsator.

Restart qucs-s.

F2 - checker error, no actions defined: nothing to do

Add DC simulation.

F2 - ERROR: Simulator crashed! Please report this error to qucs-bugs@lists.sourceforge.net

Same problem with 2N2222A from Component library.

Diff says that library files in /usr/local/share/qucs/library and /usr/local/share/qucs-s/library folders are identical.

ISSUE

Can models be created in qucs-s-0.0.22 using devices from the Component library that will run using qucsator as the Default simulator?

ra3xdh, 2020-02-02:
It is not a bug. Add "DC simulation" component to your circuit, and everything should work. For AC analysis Qucsator requires explicit DC simulation component to perform operation point analysis before execution of AC. SPICE simulators can perform OP analysis automatically and don't require the "DC sinmulation component"

G3VNC, 2020-02-06:
Sorry, I forgot that, but adding a "DC simulation" component to the circuit did not fix the problem.

Removed all qucs and qucs-s components then installed latest qucs (and qucsator) from git.

qucsator 0.0.19 seems much more tolerant of Component library devices.  I am not seeing so many crashes.

Running with the 0.0.22 AppImage; very convenient, thank you.

73

zcx2020, 2020-05-11:
This appears to be a qucsator version issue.  Cleaned out old qucs-0.0.18 from /usr/local. Compiled qucs-0.0.20-rc1 and installed into /usr/local.  qucs-s-0.0.22 now runs qucsator simulations correctly.

closed 2020-05-11 by zcx2020