ra3xdh/qucs_s #38: QUCS , PTM models

gontiavasu, 2020-05-04:
Hello People,

I have a doubt regarding QUCS, How to include PTM (available models with realistic parameters, http://ptm.asu.edu/latest.html ) models into circuits?
I am trying to download FINFET models and simulating IV characteristics of the transistor.

any help related to this will be appreciated. Thank you in advance.

ra3xdh, 2020-05-10:
There exists a special component that emulates SPICE `.INCLUDE` directive. Use this component from Spice Specific Sections tab. Then use SPICE MOSFET devices (red ones, not Qucs blue ones) and specify desired model and parameters. The PTM models should be supported by XYCE simulator, Ngspice will work. You will need to download and install XYCE in your system manually. You may require to edit the modelcard to provide XYCE compatibity, at least the model level should be replaced. Refer to XYCE documentation. The attached example illustrates how to use models in Qucs-S. 
[image]

closed 2020-05-10 by ra3xdh

gontiavasu, 2020-05-13:
Hello,

thank you so much for your inputs. I am trying to simulate it using ngspice as default simulator in QUCS studio, but not it is not simulating as fatal error is coming so do I need to install XYCE also with ngspice or only ngspice is fine?