ra3xdh/qucs_s #39: qucs-s subcircuit problem

zcx2020, 2020-05-09:
I am trying to model a broadband amplifier using an old Plessey SL560 device..  

Linux Mint 19.3 amd64, Qucs-S-0.0.22_x86_64.AppImage, qucsator-0.0.20 (from qucs-0.0.20-rc1 installation).

The type of BJTs is unspecified in the SL560 datasheet. I have chosen to use a 2N5179 model embedded in a subcircuit.

There are 3 BJTs in the SL560.  Using 3 instances of the 2N5179 in a new schematic works well.  I am getting values for voltage gain and input impedance which closely match real life.

So I thought I would embed the new SL560 schematic into a subcircuit, but the subcircuit does not work.

I get these errors when I press F2...

- Format Error: Wrong 'component' line format!  
- ERROR: Cannot load subcircuit "sl560c.sch".
- ERROR: Cannot simulate a text file!

Not sure where to go from here; any assistance would be greatly appreciated.

ra3xdh, 2020-05-10:
I cannot reproduce your issue. Please provide more details and schematic files. This error means that Qucs cannot find the subcircuit file. I see relative file paths in logs. Are you using projects to save your schematic? When using projects, Qucs saves the relative paths for subcircuit. When the schematic is opened outside the project, Qucs cannot process the relative path. Please check if you are using projects and if you are opening files correctly. Make sure that subcircuit holds an absolute path.

zcx2020, 2020-05-10:
Found the problem; the subcircuit file "sl560c.sch" got corrupted somehow,. Rebuilt from scratch it works fine.  However I do not know how the file got corrupted.

> Please check if you are using projects and if you are opening files correctly.

I believe so.  All the files exist under one project ~/.qucs/QS_SL560C_prj.

>  Make sure that subcircuit holds an absolute path.

I do not understand how to do this. 

Please see attached files.
[QS_SL560C_prj.tar.gz](https://github.com/ra3xdh/qucs_s/files/4605792/QS_SL560C_prj.tar.gz)
[image]

ra3xdh, 2020-05-10:
> Make sure that subcircuit holds an absolute path.

Yes, now I see that you are using the projects. The absolute paths are needed when you are using the subcircuit outside the projects. 

> the subcircuit file "sl560c.sch" got corrupted somehow
Is the `sl560c (backup copy).sch` the corrupted file?

zcx2020, 2020-05-10:
In past projects I have often copied subcircuit files from other projects manually i.e.

$ cp ~/.qucs/project_a/subcircuit.sch ~/.qucs/project_b

If I copy sl560c.sch to a new project, where do I need to look to make sure the copy has the right path?

Sorry, I binned the corrupted file.  The backup is a good copy.  If it happens again I will post the corrupt file here.

closed 2020-05-11 by zcx2020