ra3xdh/qucs_s #40: Equation and Plots

edson-acordi, 2020-05-15:
Hi, first, I would like to congrats you for the great software!
I am trying to calculate and view the RMS and AVG value of a sinusoid waveform (va => given by a nodename).
I have already tried all the below equations:
rms(va.v)
rms(va)
rms(va.v, time)

I have used only "Transient Simulation".

Moreover, to view the result, I tried to use tabular/cartesian diagrams, however without success.
There is some trick to do it?

ra3xdh, 2020-05-16:
RMS calculation in the Ngspice is implemented using the .MEASURE command which is not implemented in GUI.  You can try to do something using the custom Nutmeg scripting. For Nutmeg scipting basics look at this example: https://qucs-s-help.readthedocs.io/en/latest/ASim.html#spectrum-analysis-with-ngspice-and-nutmeg-scripting Unfortunately a more detailed there is no more detailed documentation on this topic for now.

edson-acordi, 2020-05-19:
Hi ra3xdh, thank you for your reply!

Ok, I could realize that!
I take the current from the current probe vpr1.
So, the current is vpr1#branch and the RMS value can be computed with a Nutmeg equation as,
y1=sqrt(avg(vpr1#branch^2))

Now, I am migrating from LTspice to Qucs-S!

Thank you and again congrats for the great FOSS software!

closed 2020-06-19 by ra3xdh