ra3xdh/qucs_s #41: F6

zcx2020, 2020-05-16:
Using qucs-s-0.0.22, qucsator-0.0.20,  ngspice revision 27, linux mint 19.3 x86_64.

If I do a qucsator simulation F6 shows the netlist for that simulation.

If I change the default simulator to ngspice, restart qucs-s and then simulate a different circuit F6 still shows the netlist for the previous qucsator simulation, not the one for the current ngspice simulation.

Is the only way to view a ngspice netlist to save the netlist and then open it with a text editor?

Is this a bug?

ra3xdh, 2020-05-16:
Yes, it is a bug. I can reproduce it. You may also open `$HOME/.qucs/spice4qucs/spice4qucs.cir` with text editor. It holds the latest simulated SPICE netlist.

zcx2020, 2020-05-16:
Thanks for the tip - I'll use a desktop launcher to display it.

ra3xdh, 2022-02-19:
Fixed with https://github.com/ra3xdh/qucs_s/commit/548a47519b299070cfc8dac8a7099b7921cb1d6f For Xyce it shows the netlists for all simulations presented in the circuit.

commit 548a4751 (2022-02-19): Show spice netlist on F6 (fixed #41)
closed 2022-02-19 by ra3xdh