ra3xdh/qucs_s #42: Q(NPN) BJT parameters

zcx2020, 2020-06-03:
Using qucs-s-0.0.22, ngspice revision 27, linux mint 19.3 x86_64.

If I place a Q(NPN) BJT and add a model to the Component Properties...

Q = Q2N5179
Q_Line 2 =
Q_Line 3 =
Q_Line 4 =
Q_Line 5 =

I can either use a .INCLUDE section to include a .lib file that contains the .model parameters, or a .MODEL section and set the parameters in the Component Properties...

Line_1 = .model Q2N5179 NPN(Is=69.28E-18 Xti=3 Eg=1.11 Vaf=100 Bf=282.1
Line_2 = + Ne=1.177 Ise=69.28E-18 Ikf=22.03m Xtb=1.5 Br=1.176 Nc=2 Isc=0 Ikr=0
Line_3 = + Rc=4 Cjc=893.1f Mjc=.3017 Vjc=.75 Fc=.5 Cje=939.8f Mje=.3453 Vje=.75
Line_4 = + Tr=1.588n Tf=141.1p Itf=.27 Vtf=10 Xtf=30 Rb=10)
Line_5 =

The netlist gets built correctly and ngspice runs.

But if I try to set the parameters in the Q(NPN) BJT Component Properties directly...

Q  = .model Q2N5179 NPN(Is=69.28E-18 Xti=3 Eg=1.11 Vaf=100 Bf=282.1
Q_Line 2 = + Ne=1.177 Ise=69.28E-18 Ikf=22.03m Xtb=1.5 Br=1.176 Nc=2 Isc=0 Ikr=0
Q_Line 3 = + Rc=4 Cjc=893.1f Mjc=.3017 Vjc=.75 Fc=.5 Cje=939.8f Mje=.3453 Vje=.75
Q_Line 4 = + Tr=1.588n Tf=141.1p Itf=.27 Vtf=10 Xtf=30 Rb=10)
Q_Line 5 =

The netlist does not get built correctly and ngspice generates errors.

Is this a bug or is it me?

tvrusso, 2020-06-03:
This is not a bug.

In SPICE, one does not attach the entire contents of a model "card" to the device instance.  It is always done by creating a named model "card" and referencing that name in the device instance.

Doing what you've done is simply generating a "Q" line in the netlist with invalid syntax.

To be concrete:  your first example would result in the following Q line ("device instance") in the netlist:

    Q1 collector base emitter Q2N5179

This tells the code to use the properties of the Q2N5179 model to get the parameters of the BJT.  The model card needed will either be taken from the .lib you say you include, or via the ".model" you tried:

     .model Q2N5179 NPN(Is=69.28E-18 Xti=3 Eg=1.11 Vaf=100 Bf=282.1
    + Ne=1.177 Ise=69.28E-18 Ikf=22.03m Xtb=1.5 Br=1.176 Nc=2 Isc=0 Ikr=0
    + Rc=4 Cjc=893.1f Mjc=.3017 Vjc=.75 Fc=.5 Cje=939.8f Mje=.3453 Vje=.75
    + Tr=1.588n Tf=141.1p Itf=.27 Vtf=10 Xtf=30 Rb=10)

This is all legal netlisting syntax.

But when you tried to add the model information to the Q component, that resulted in a netlist:
    Q1 collector base emitter .model Q2N5179 NPN(Is=69.28E-18 Xti=3 Eg=1.11 Vaf=100 Bf=282.1
    + Ne=1.177 Ise=69.28E-18 Ikf=22.03m Xtb=1.5 Br=1.176 Nc=2 Isc=0 Ikr=0
    + Rc=4 Cjc=893.1f Mjc=.3017 Vjc=.75 Fc=.5 Cje=939.8f Mje=.3453 Vje=.75
    + Tr=1.588n Tf=141.1p Itf=.27 Vtf=10 Xtf=30 Rb=10)

which is an invalid Q line.  Don't do that.

zcx2020, 2020-06-04:
> This is not a bug.
> 
> In SPICE, one does not attach the entire contents of a model "card" to the device instance. It is always done by creating a named model "card" and referencing that name in the device instance.

Understood,  but the Q(NPN) BJT Component Properties allow it.  Maybe there is a good reason for that.

[image]

> But when you tried to add the model information to the Q component, that resulted in a netlist:
>
> ```
> Q1 collector base emitter .model Q2N5179 NPN(Is=69.28E-18 Xti=3 Eg=1.11 Vaf=100 Bf=282.1
> + [...]

Yep.  That's what I get.

As a long time qucs user I am trying to migrate away from qucsator to ngspice.  Still have a lot to learn.

Thanks for the explanation.

tvrusso, 2020-06-04:
You are allowed to have multiple lines of BJT *instance* parameters in the component.  There's a difference.

Spice device lines support parameters that are specific to that instance of the device, and model parameters that are meant more to define a class of devices.  You associate the model with the instance as you have, but then there can be additional parameters that tune the instance.

 BJTs don't have that many instance parameters, but they do have a few.  For example, the SPICE default BJT has an area parameter, and OFF or ON parameter (basically giving the state of the transistor at the operating point, usually only to help convergence), a temperature, and initial conditions (again to help convergence of the operating point).  You can specify all of those on the Q_Line_n lines with continuation characters.

But the model parameters like is, xti, etc. must be on a separate model card, and are referenced by putting the model name on the first Q line (it needs to be first).

Just the way SPICE works.  instance and model parameters are very different and need to be specified in different places.

spice3F5, ngspice, and Xyce are all that way.

zcx2020, 2020-06-04:
OK.  I've got it now.

Thanks again for explaining.

closed 2020-06-04 by zcx2020