ra3xdh/qucs_s #43: ngspice transient simulations

zcx2020, 2020-06-04:
Using qucs-s-0.0.22, ngspice revision 27, linux mint 19.3 x86_64.

Although transient simulations are a lot faster using ngspice than qucsator, plotting them in qucs-s seems hit and miss.

As an example I have two .sch files with just an AC voltage source and a resistor.  They have been pared back from two larger .sch files.  Although they look identical test_trans_01,sch plots OK, but test_trans_02.sch does not.

ngspice simulates both files with identical output and no errors.

Please see attached netlists (.txt extensions for this forum).

[test_trans_01.cir.txt](https://github.com/ra3xdh/qucs_s/files/4730455/test_trans_01.cir.txt)

[test_trans_02.cir.txt](https://github.com/ra3xdh/qucs_s/files/4730458/test_trans_02.cir.txt)

The only differences I can see  between the two netlists are reversed in the following two lines are reversed in test_trans_02.cir...

V2 Vsrc 0 DC 0 SIN(0 0.0075 10MEG 0 0 0) AC 0.0075
R1 0 Vsrc  1MEG

and 

 test_trans_01.cir contains the line...
write test_trans_01_tran.txt v(Vsrc) 

whilst test_trans_02.contains the line...
write test_trans_02_tran.txt v(Vsrc)

The test_trans_0#.txt files are not on the system.

Because ngspice runs OK with no errors, ISTM that the problem must be in qucs-s?

zcx2020, 2020-06-04:
Found it.  Source frequency must be 10 MHz or 10 Meg, not 10M.

closed 2020-06-04 by zcx2020

tvrusso, 2020-06-04:
Be aware that in SPICE a lot of these things are very counter-intuitive:  spice netlists are case-insensitive, so "m" and "M" (while humans think of these as milli- and mega-) mean the same thing: 1e-3.  As a result, you must use the three-character suffix "MEG" to get 1e6 in the netlist (or just write 10e6 where you want 10 meg).

Spice also ignores units.  "10MHz" in SPICE means "10e-3".  It doesn't care if you write "10MHz" or "10MV" --- it still means 10e-3, and the Hz or V at the end is ignored (think of it as a comment).

Qucs-S does some conversion between how Qucs treats units and multipliers, but one must watch it when you want mega-anything, and verify that the netlist it creates has MEG where you mean 1e6.

zcx2020, 2020-06-05:
Should have known.  Already been tripped up by the 1M vs 1Meg thing for resistors.

Will RTM some more.  Is this as good as any?

http://ngspice.sourceforge.net/docs/ngspice-manual.pdf

tvrusso, 2020-06-05:
I'm not an ngspice user, but i would assume that the ngspice manual is a great resource for using that tool :).