ra3xdh/qucs_s #47: F5

zcx2020, 2020-06-13:
Let's say there are two schematics open in two tabs.

The first schematic has a problem which prevents ngspice from producing any useful results.

Press F2 and the Statusbar says Ngspice Warnings in last simulation! Press F5.  Correct.

The second schematic has no problem and is simulated without error.  If I switch to the second tab the Statusbar still displays the warning from the first tab.  Incorrect.

Now press F2 and the Statusbar says Ngspice no warnings.  Correct.
 
Switch back to the first tab and the Statusbar still says Ngspice no warnings.  Incorrect.

Should the warning message and log.txt somehow be local to each tab and preserved if another tab is simulated, so that F5 works correctly across tabs?

ra3xdh, 2020-06-19:
This behavior is inherited from the basic Qucs project. In principle it could be rewritten in a following manner: every log file is connected to the specific schematic file, but not in the near future.

zcx2020, 2020-06-20:
Understood.  Thanks anyway.

closed 2020-06-20 by zcx2020