ra3xdh/qucs_s #48: spurious plot lines

zcx2020, 2020-06-15:
Linux Mint 19.3 x86_64, qucs-s-0.0.22, ngspice revision 27.

Not sure if this is a qucs-s or ngspice issue.  Anyone else seen this?

[image]

ra3xdh, 2020-06-19:
I cannot reproduce this bug. Could you post a schematic (`*.sch` file) somewhere?

zcx2020, 2020-06-20:
Yes...

[test_2N2222_01.sch.txt](https://github.com/ra3xdh/qucs_s/files/4807723/test_2N2222_01.sch.txt)

ra3xdh, 2023-07-31:
I decided to investigate this issue. I was able to reproduce it, but it is not a bug. The problem is related to the `i` variable name which is reserved for imaginary unit `sqrt(-1)`. Ngspice doesn't show the warning message about variable redefinition. The simulation works as expected after renaming this variable. Closing this.

[image]