ra3xdh/qucs_s #49: Version String Missing in .dat Files in Version 0.0.22

cweickhmann, 2020-06-18:
I observed that in 0.0.22, the `<Qucs Dataset >` tag is not complete. In particular, it is missing the version string (in this case 0.0.22) which makes e.g. `qucsconv` refuse working.

**MWE**
* Create file (here `single_coil.sch`)
* Place any simple setup in there (in my case: DC source, resistor, GND, connect them, add a DC Simulation)
* Check the `single_coil.dat` file.

In 0.0.19, the first line reads
```
<Qucs Dataset 0.0.19>
```

In 0.0.22, the first line reads
```
<Qucs Dataset >
```

This makes qucsconv hickup and refuse to work in a on first sight rather cryptic error message (`line 1: syntax error, unexpected '<', expecting Version`).

ra3xdh, 2020-06-19:
Please clarify. Qucs-S cannot produce the `*.dat` files. It only converts the Ngspice/Xyce datasets into the `*.dat.ngspice` or `*.data.xyce` files. Did you mean these files? The `*.dat` files could only come from qucsator simulation engine which is not a part of the Qucs-S package.

cweickhmann, 2020-06-20:
Indeed, this happens using `qucsator`. So, I do mean the `.dat` file, neither ngspice nor xyce involved.
The reason I posted this here is because it does not happen with the same combination on standard qucs.
Any idea why qucsator appears to write an empty version string?

closed 2022-03-20 by ra3xdh