ra3xdh/qucs_s #51: PoC Flatpak packaging

tinywrkb, 2020-07-03:
[This is an initial Flapak manifest for Qucs-S](https://github.com/tinywrkb/flatpaks/blob/master/com.github.ra3xdh.qucs-s/com.github.ra3xdh.qucs-s.yaml), not really tested except a basic simulation.  
It definitely needs more work, testing, and cleanup.  
This needs [Qucs base app](https://github.com/tinywrkb/flatpaks/blob/master/com.github.qucs.qucs.BaseApp/com.github.qucs.qucs.BaseApp.yaml) when building the app, so build and install it first.

I've followed the Arch Linux packaging and added every optional dependency for Qucs-S and its dependencies. So you have Ngspice, FreeHDL, GTKWave, and even also Qucs packaged.

I don't plan to publish this on Flathub myself, but I hope this might encourage you to choose Flatpak as a distribution channel.

BTW Qucs is also packaged [here](https://github.com/tinywrkb/flatpaks/blob/master/com.github.qucs.qucs/com.github.qucs.qucs.yaml).

ra3xdh, 2022-05-09:
I have recently started the generation of the Flatpak manifest for Qucs-S. I didn't find a way to launch an external application from system `/usr` from the flatpak. Therefore I decided to include the ngspice into the Flatpak package. The qucsator has dependency on ADMS that depends on Perl modules. And therefore the adding of the qucsator into the Flatpak is not a straightforward task. So I decided to restrict the Flatpak version of the Qucs-S by Ngspice support only. The manifest draft could be found here: https://github.com/ra3xdh/qucs_scripts/blob/master/FLATPAK/com.github.ra3xdh.qucs-s.yaml

tinywrkb, 2022-05-09:
> The qucsator has dependency on ADMS that depends on Perl modules. And therefore the adding of the qucsator into the Flatpak is not a straightforward task.

@ra3xdh you can use [flatpak-cpan-generator.pl](https://github.com/flatpak/flatpak-builder-tools/tree/master/cpan).

I packaged ADMS with QUCS like this
```
- name: adms
  config-opts:
   - --enable-maintainer-mode
   - PERL5LIB=/app/lib/perl5
  make-args:
   - PERL5LIB=/app/lib/perl5
  sources:
   - type: archive
     url: https://github.com/Qucs/ADMS/archive/release-2.3.7.tar.gz
     sha256: 0d24f645d7ce0daa447af1b0cff1123047f3b73cc41cf403650f469721f95173
  modules:
    - name: perl-dependencies
      buildsystem: simple
      build-options:
        env:
          - PERL5LIB=/app/lib/perl5
      build-commands:
        - perl-libs/install.sh
      post-install:
        # workaround for cpanm acting silly and install files without write permission
        - find /app -type f ! -perm -0200 -exec
            chmod u+w '{}' \;
      sources:
        - perl-dependencies-sources.json
        - type: shell
          commands:
            # fix the generated install.sh to avoid having it try to install into /usr
            - sed -e 's|\(perl Makefile.PL\) \&\&|\1 INSTALL_BASE=/app \&\&|'
                -e 's|\(perl Build.PL\) \&\&|\1 --install_base=/app \&\&|'
                -i perl-libs/install.sh
```

And `perl-dependencies-sources.json` was generated by running

```
$ flatpak-cpan-generator.pl --output perl-dependencies-sources.json XML::LibXML
```

tinywrkb, 2022-05-09:
```
- --socket=x11
```
Unless you're forcing XCB QPA platform in the application or using ngspice's X11 plotting, it should have instead
```
--socket=fallback-x11
--socket=wayland
```
Please read the [standard permission section](https://docs.flatpak.org/en/latest/sandbox-permissions.html#standard-permissions).

```
  - name: ngspice
    build-options:
        strip: true
```
This means that you're throwing away ngspice's symbols, instead of extracting them and moving into a separate `.Debug` extension.  
In fact, due to `flatpak-builder` bug, this actually remove symbols from everything that is already in `$FLATPAK_DEST`.

And last, this is obviously missing an AppStream MetaInfo, which will be needed for Flathub submission. Please see the [documentation here](https://freedesktop.org/software/appstream/docs/chap-Metadata.html), the [AppStream MetaInfo Creator](https://www.freedesktop.org/software/appstream/metainfocreator/#/), and the [Appstream section in App Requirements wiki page](https://github.com/flathub/flathub/wiki/App-Requirements#appstream).

tinywrkb, 2022-05-12:
I updated the example `ADMS` packaging to work with the latest revision of `latpak-cpan-generator.pl`.

luzpaz, 2023-08-31:
Is flatpak qucs-s support up to date ?

ra3xdh, 2024-02-12:
> Is flatpak qucs-s support up to date ?

The main issue with Flatpak was sandboxing concept which is against Qucs-S architecture. Qucs-S is intended to reference an external simulation kernel. It is required to compile every simulation kernel and provide fixed versions inside flatpak. The issue may be partially resolved after implementing Qucsator_RF integration. Otherwise supported kernels may be restricted to Ngspice only. Xyce build with flatpak is not possible because of Trillinos libraries.

NN708, 2024-09-06:
Hello @ra3xdh ,

Would it be possible to submit the current manifest to Flathub? I believe that even with support for only Ngspice, it would still be quite valuable for users, especially on immutable systems.

ra3xdh, 2024-09-07:
> Would it be possible to submit the current manifest to Flathub?

The following things should be corrected in the existing manifest: 

* Hardcoded qucs-s tarball version
* Hardcoded Ngspice version
* Qucs-S repository should be packaged with submodules.

It's need to learn if Flatpack supports automatic usage of the latest stable version.  Also the isolation concept doesn't allow to use external Ngspice or Xyce. 

> I believe that even with support for only Ngspice, it would still be quite valuable for users

The QucsatorRF is a part of Qucs-S package since v24.2.0 It doesn't depend on Perl anymore. See #370 So two simulators should be available.

NN708, 2024-09-07:
Thanks for your response! It's exciting to hear that more simulation kernels could be supported. While the isolation in Flatpak can be a bit of a hassle due to the need to package all dependencies, it also means users can simply install the Flatpak package to get a fully functional Qucs-S, without having to install additional packages on their host system.

> It's need to learn if Flatpak supports automatic usage of the latest stable version.

IMHO, we should hardcode Qucs-S and its dependencies in the manifest. Flathub provides an [External Data Checker](https://docs.flathub.org/docs/for-app-authors/external-data-checker) that can automatically check for dependency updates and create PRs for the manifest.

ra3xdh, 2024-09-15:
I will try to prepare Flatpak manifest update by the next release. Contribution is welcome. The manifest files could be placed in `contrib` subdirectory of the source tree.

NN708, 2024-11-10:
Hi @ra3xdh,

I've successfully updated the manifest to include all the latest changes: [io.github.ra3xdh.qucs_s.yml](https://github.com/NN708/flathub/blob/qucs-s/io.github.ra3xdh.qucs_s.yml). It now automatically detects both Ngspice and Qucsator. Since the manifest is typically submitted to a dedicated repository under [Flathub](https://github.com/flathub/), would it be alright for me to submit it there?

Additionally, as @tinywrkb mentioned, we need to add an AppStream MetaInfo to this repository as well:

> And last, this is obviously missing an AppStream MetaInfo, which will be needed for Flathub submission. Please see the [documentation [...]

ra3xdh, 2024-11-10:
> Since the manifest is typically submitted to a dedicated repository under [Flathub](https://github.com/flathub/), would it be alright for me to submit it there?

Thanks! Yes, you may submit the manifest. The QucsatorRF is a part of Qucs-S package, so only two repositories (Qucs-S and Ngspice) are required.

NN708, 2024-11-12:
Since Flathub only accepts stable versions, we are currently waiting for a patch release to proceed with our submission to Flathub. The release section of [MetaInfo](https://github.com/NN708/qucs_s/blob/current/contrib/io.github.ra3xdh.qucs_s.metainfo.xml) should be updated before the next release following the [MetaInfo guidelines](https://docs.flathub.org/docs/for-app-authors/metainfo-guidelines/#release).

ra3xdh, 2024-11-12:
I am planning to prepare v24.4.1 release by the end of this week. This will include some bugfixes implemented recently.

ra3xdh, 2024-11-14:
I have updated MetaInfo in #1061

ra3xdh, 2024-11-14:
I have just uploaded the 24.4.1 release packages. It contains the MetaInfo now.

ra3xdh, 2024-11-16:
@NN708 Thanks for adding Flatpak support! This issue may be closed as resolved now.

closed 2024-11-16 by ra3xdh

ra3xdh, 2024-11-17:
I have made an announcement #1069