ra3xdh/qucs_s #53: Problem with Paramter Sweep of multiple attribute devices

HB9GKW, 2020-08-14:
I'm having problems with performing a parameter sweep of devices with multiple attributes like an ideal AC voltage source. Sweeping devices with single attribute, e.g. DC voltage source, R, C, L.. works fine. I also tried to put an AC voltage source inside a sub-circuit having only one variable attribute to call from main circuit without success (Param = @SUB[VAC]). Writing a Nutmeg script by looping the command "alter @vg[sin] [ -1 $&VAC 100MEG ]  $ vector" seems to be an alternative but I failed in getting the simulation results back in a useful way to plot. Is there any trick in how this could work out?

ra3xdh, 2020-08-14:
Yes, the Qucs-S has problems with sweeping device parameters defined as vectors like for SIN source. Currently it could not be resolved automatically.  The only solution is to use Nutmeg scritping. Look at this example https://github.com/ra3xdh/qucs/issues/8#issuecomment-254825825 You need to enter variables to plot like `v(n1)` manually in the Custom simulation properties dialog in the "Variables to plot" input field.

HB9GKW, 2020-08-15:
Thanks for this very useful link! I didn't enter the variables to plot in the properties dialog. Now it works! 73s!

HB9GKW, 2020-08-18:
Thanks to your advice the parameter sweep of vector devices works now but unfortunately I cannot display the scalar max value of a transient analysis versus the swept value. Lets assume sweeping the RF input voltage of a simple RF detector and plotting the rectified output voltage v(out) versus input amplitude VAC_act. I tried to adapt your Nutmeg script to:

let start_VAC=1
let stop_VAC=5
let delta_VAC=1

let VAC_act=start_VAC
while VAC_act le stop_VAC
   alter @V1[SIN] [ 0 $&VAC_act 0 0 ]
   tran 0.1n 100n
   let vmax=maximum(v(out))
   write VAC_sweep.out vmax
   set appendwrite
   let VAC_act=VAC_act+delta_VAC
end

vmax still has time dependency with but calculates the correct value as first vector element. The rest however gets filled with 0 and therefore cannot be plotted versus VAC_act.

Using ngspice command .measure tran vmax MAX v(out) from=10n to=100n inside the while loop did not write to vmax at all.. 
Is there a chance in plotting vmax@VAC_act?

ra3xdh, 2022-02-19:
> vmax still has time dependency with but calculates the correct value as first vector element. The rest however gets filled with 0 and therefore cannot be plotted versus VAC_act.

Sorry for the delayed answer. The vmax is the single value and the vector will be filled with zeros except the first element. There is no error. You need to write the vmax to another output or use the table output to plot it. Please refer to Ngspice manual.

closed 2022-02-19 by ra3xdh