ra3xdh/qucs_s #54: qucsator and unforking

felix-salfelder, 2020-09-29:
(Not an issue, but an announcement)

I have moved Qucsator into a separate repository within the Qucs project. It should be shared between qucs-s and qucs.

You will need the equivalent of https://github.com/Qucs/qucs/pull/1015, and perhaps check if there are any deviations that need to be considered. The rest of Qucs-s should be ported back to Qucs once Qt5 is working sufficiently.

ra3xdh, 2022-03-20:
Hello Felix,

The Qucs-S doesn't contain qucsator by itself. So, the Qucsator could be installed in the system independently and shared by several frontends. I have recently evaluated the `modular` branch of the Qucs project and I don't consider the unforking in the near future. The `modular` branch still be in very early development stage and has many incompatibilities with Qucs-S. And it's not clear for me how to resolve it. Currently the software architecture of Qucs-S is monolithic and full of legacy code and may be far from ideal, but it is stable and ready for the practical usage. I cannot say this about the `modular` branch. 

Best regards,
Vadim

closed 2022-03-20 by ra3xdh

felix-salfelder, 2022-03-21:
On Sun, Mar 20, 2022 at 05:27:35AM -0700, Vadim Kusnetsov wrote:
> The Qucs-S doesn't contain qucsator by itself. So, the Qucsator could
> be installed in the system independently and shared by several
> frontends.

Thanks. The latest Qucsator release has critical bugs that make it
unusable on some platforms. A maintainer is required to eventully leave
a functioning program to posterity.

> The `modular` branch [..] has many incompatibilities with Qucs-S. And
> it's not clear for me how to resolve it.

The starting point is a spice netlister and a (ng?) spice driver. This
will be very similar to the qucsator netlister and qucsator driver (with
some executable name and syntax replaced respectively). Possible next
step will be a "raw" file parser plugin. Currently, there are plugins
for reading and writing .dat files only -- but they give the idea.

Whoever does it will receive help and probably learn quite a lot.