ra3xdh/qucs_s #56: Transient simulation parameter sweep outputting garbage

Compizfox, 2020-11-19:
I am using Qucs-S + ngspice to simulate the transient response of a current source circuit. I want to sweep a parameter (the value of a resistor), but when that parameter sweep includes more than 2 points, the results are garbage (ridiculously high/low anomalous values).

When inputting these values for the swept parameter manually (without the Parameter Sweep box), or when using the Parameter Sweep box with only 1 or 2 points, the results are as expected.

I don't understand what's going on here... The schematic is as follows:

```
<Qucs Schematic 0.0.22>
<Properties>
  <View=128,39,1155,677,1.48538,0,52>
  <Grid=10,10,1>
  <DataSet=test2.dat>
  <DataDisplay=test2.dpl>
  <OpenDisplay=1>
  <Script=test2.m>
  <RunScript=0>
  <showFrame=0>
  <FrameText0=Title>
  <FrameText1=Drawn By:>
  <FrameText2=Date:>
  <FrameText3=Revision:>
</Properties>
<Symbol>
</Symbol>
<Components>
  <VProbe Vds 1 970 420 -16 28 0 3>
  <Lib T1 1 860 420 8 -26 0 0 "NMOSFETs" 0 "IRLZ34N" 0>
  <Diode D_1N4007_1 5 960 300 -13 -26 0 3 "76.9p" 0 "1.45" 0 "26.5p" 0 "0.333" 0 "0.7" 0 "0.5" 0 "0" 0 "0" 0 "2" 0 "42m" 0 "4.32u" 0 "0" 0 "0" 0 "1" 0 "1" 0 "1k" 0 "5u" 0 "26.85" 0 "3.0" 0 "1.11" 0 "0.0" 0 "0.0" 0 "0.0" 0 "0.0" 0 "0.0" 0 "0.0" 0 "26.85" 0 "1.0" 0 "normal" 0>
  <IProbe Ifw 1 910 360 -26 16 0 0>
  <VProbe Vsol 1 1080 300 -16 28 0 3>
  <L L1 1 860 320 10 -26 0 1 "100 mH" 1 "" 0>
  <Vdc V3 1 860 200 -63 -26 0 3 "12 V" 1>
  <GND * 1 860 170 0 0 0 2>
  <IProbe Ifb 1 800 360 -26 -35 0 2>
  <IProbe Is 1 780 530 -26 -35 0 2>
  <GND * 1 640 320 0 0 0 2>
  <Vdc V4 1 640 350 -63 -26 0 3 "5V" 1>
  <GND * 1 640 460 0 0 0 0>
  <R R4 1 480 460 15 -26 0 1 "330 Ohm" 1 "26.85" 0 "0.0" 0 "0.0" 0 "26.85" 0 "US" 0>
  <R R5 1 540 530 -26 -53 0 2 "1 Ohm" 1 "26.85" 0 "0.0" 0 "0.0" 0 "26.85" 0 "US" 0>
  <GND * 1 180 530 0 0 0 0>
  <R R1 1 300 400 -26 15 0 0 "4700 Ohm" 1 "26.85" 0 "0.0" 0 "0.0" 0 "26.85" 0 "US" 0>
  <Vrect V1 1 190 460 18 -26 0 1 "5 V" 0 "2 ms" 0 "2 ms" 0 "1 us" 0 "1 us" 0 "0 ns" 0>
  <C C2 1 380 460 17 -26 0 1 "10 uF" 1 "" 0 "neutral" 0>
  <Lib OP1 1 630 420 40 34 0 0 "SpiceOpamp" 0 "LM358" 0>
  <IProbe g 1 800 420 -26 16 0 0>
  <R R8 1 720 420 -26 15 0 0 "0 Ohm" 1 "26.85" 0 "0.0" 0 "0.0" 0 "26.85" 0 "US" 0>
  <R R7 1 770 390 15 -26 0 1 "1000 Ohm" 1 "26.85" 0 "0.0" 0 "0.0" 0 "26.85" 0 "US" 0>
  <R R6 1 860 260 15 -26 0 1 "0" 1 "26.85" 0 "0.0" 0 "0.0" 0 "26.85" 0 "US" 0>
  <.SW SW1 1 440 190 0 59 0 0 "TR1" 1 "list" 1 "R6" 1 "30" 0 "40" 0 "[30; 31, 32]" 1 "false" 0>
  <.TR TR1 1 210 200 0 59 0 0 "lin" 1 "0" 1 "25 ms" 1 "1000" 0 "Trapezoidal" 0 "2" 0 "1 ns" 0 "1e-16" 0 "150" 0 "0.001" 0 "1 pA" 0 "1 uV" 0 "26.85" 0 "1e-3" 0 "1e-6" 0 "1" 0 "CroutLU" 0 "no" 0 "yes" 0 "0.5 us" 0>
</Components>
<Wires>
  <860 450 950 450 "" 0 0 0 "">
  <950 430 950 450 "" 0 0 0 "">
  <860 390 950 390 "" 0 0 0 "">
  <950 390 950 410 "" 0 0 0 "">
  <860 350 860 360 "" 0 0 0 "">
  <860 360 860 390 "" 0 0 0 "">
  <860 360 880 360 "" 0 0 0 "">
  <960 330 960 360 "" 0 0 0 "">
  <940 360 960 360 "" 0 0 0 "">
  <960 360 1060 360 "" 0 0 0 "">
  <1060 310 1060 360 "" 0 0 0 "">
  <860 220 860 230 "" 0 0 0 "">
  <960 230 960 270 "" 0 0 0 "">
  <860 230 960 230 "" 0 0 0 "">
  <1060 230 1060 290 "" 0 0 0 "">
  <960 230 1060 230 "" 0 0 0 "">
  <830 360 860 360 "" 0 0 0 "">
  <860 450 860 530 "" 0 0 0 "">
  <810 530 860 530 "" 0 0 0 "">
  <330 400 380 400 "" 0 0 0 "">
  <380 400 380 430 "" 0 0 0 "">
  <380 400 480 400 "" 0 0 0 "">
  <480 400 480 430 "" 0 0 0 "">
  <190 400 270 400 "Vctrl" 280 370 67 "">
  <190 400 190 430 "" 0 0 0 "">
  <570 530 590 530 "" 0 0 0 "">
  <480 490 480 530 "" 0 0 0 "">
  <180 530 190 530 "" 0 0 0 "">
  <190 490 190 530 "" 0 0 0 "">
  <480 530 510 530 "" 0 0 0 "">
  <190 530 380 530 "" 0 0 0 "">
  <380 530 480 530 "" 0 0 0 "">
  <380 490 380 530 "" 0 0 0 "">
  <590 530 750 530 "" 0 0 0 "">
  <590 440 590 530 "Vinrev" 620 500 69 "">
  <480 400 590 400 "Vin" 570 370 91 "">
  <750 420 770 420 "Vout" 770 430 11 "">
</Wires>
<Diagrams>
</Diagrams>
<Paintings>
</Paintings>
```

Let me know if you need more info.

ra3xdh, 2020-11-22:
Hello, It is the fundamental limitation of the Qucs dataset format. For Ngspice every transient simulation may have a different number of time steps. When such simulation are combined into a single dataset, the data may become unreadable, beacuse Qucs plotting engine expects the same data points number for every parameter sweep step. It cannot be fixed without the breaking of the backward compatibility of Qucs XML datasets.

closed 2020-11-22 by ra3xdh

Compizfox, 2020-11-23:
Thanks for the explanation.

Just one question: shouldn't it be possible to avoid this issue by making sure that the timestep is fixed? I tried doing this by setting `MinStep` and `MaxStep` in the transient simulation properties to the same value, but no luck.