ra3xdh/qucs_s #57: Crash (segfault) on adding any device

auxym, 2020-12-03:
##What  I'm doing:

1. Open qucs-s
2. Click any device from the components library (eg a resistor)
3. Click on the canvas to place it

## What happens

At this point qucs-s crashes without any other error. If I started it in a terminal, I get the following output:

```
Unable to load library icui18n "Cannot load library icui18n: (icui18n: cannot open shared object file: No such file or directory)" 
Debug: pressed CompComps id 1 "Resistor US" 
Debug: slotSelectComponent() 0 1 
Debug: mod->info true 
Debug: mod->infoVA false 
fish: “qucs-s” terminated by signal SIGSEGV (Address boundary error)
```

## My system

Running on manjaro linux. Qucs-s was compiled and installed using the AUR package: https://aur.archlinux.org/packages/qucs-s/ (v 0.0.22-1 at the time of writing).

Please let me  know if I can give any other information.

Mazurel, 2020-12-03:
I think it may be due to qt4 missing some dependecy (apparently translation one), you can try installing `icu` package as it provides the missing library and it may fix the issue.
I have tested it on my machine with Nixos and can't really replicate this problem so I guess it may be Arch specific issue.

auxym, 2020-12-03:
Odd, as I do have `icu` package installed. Might be an arch-specific library naming thing. I left a comment on the AUR page, hopefully the maintainer or someone else more knowledgeable than I has an idea.

auxym, 2020-12-04:
Additional bit of info: I tried the AppImage. It works (no segfault) which is good :) It still gives the icui18n error, which leads me to believe that the segfaults in the AUR version might not be related to icui18n.

Mazurel, 2020-12-04:
Do you have the same issue with Qucs ?

auxym, 2020-12-08:
I just installed qucs (built from the AUR,  again). In fact, yes, I'm getting the exact same error in qucs. I guess I should report the bug to qucs then?

Mazurel, 2020-12-08:
Yeah, that is what I would suggest. If you will get any response you can mention it here, it would help a lot.