ra3xdh/qucs_s #60: Data display in schematics created by Qucs (not -S)

patrickschulz, 2021-02-09:
Hi, I've tested Qucs-S today for the first time since I've had some problems with Qucs Transient simulation. Therefore, I've used a schematic that I created in Qucs. The simulation worked well, but I could not select ngspice as source for the data in the diagram.

ra3xdh, 2021-02-09:
The desired simulator should be selected from the drop-down list in the diagram properties window. What Qucs version are you using? Could you send me a schematic to reproduce the problem?

patrickschulz, 2021-02-09:
I'm using Qucs 0.0.20 and Qucs-s 0.0.22. While looking for the schematic file for your request I've noticed that the names of the dataset, script and display files where different from the main schematic name (this happened because I've accidently named the schematic incorrectly and renamed it afterwards). Changing this to the correct name fixed the issue, I can now select ngspice as data source. The weird thing is that the data file (main.dat.ngspice) was generated with the data, but it did not get loaded correctly somehow. Long story short, here is the UNMODIFIED schematic which should produce the error:
<Qucs Schematic 0.0.22>
<Properties>
  <View=-14,90,970,756,1.3831,0,0>
  <Grid=10,10,1>
  <DataSet=mai.dat>
  <DataDisplay=mai.dpl>
  <OpenDisplay=0>
  <Script=mai.m>
  <RunScript=0>
  <showFrame=0>
  <FrameText0=Title>
  <FrameText1=Drawn By:>
  <FrameText2=Date:>
  <FrameText3=Revision:>
</Properties>
<Symbol>
  <.ID -20 -16 SUB>
  <Line -20 20 40 0 #000080 2 1>
  <Line 20 20 0 -40 #000080 2 1>
  <Line -20 -20 40 0 #000080 2 1>
  <Line -20 20 0 -40 #000080 2 1>
</Symbol>
<Components>
  <GND * 5 160 310 0 0 0 0>
  <GND *1 5 40 310 0 0 0 0>
  <Vdc V1 1 160 280 18 -26 0 1 "0.9" 1>
  <Vdc V2 1 40 280 18 -26 0 1 "0" 1>
  <GND * 5 340 340 0 0 0 0>
  <MOSFET T2 1 560 220 34 -26 0 0 "pfet" 0 "-300e-3" 1 "2e-5" 1 "0.0" 0 "0.6 V" 0 "0.0" 1 "0.0 Ohm" 0 "0.0 Ohm" 0 "0.0 Ohm" 0 "1e-14 A" 0 "1.0" 0 "1 um" 0 "1 um" 0 "0.0" 0 "0.1 um" 0 "0.0" 0 "0.0" 0 "0.0" 0 "0.0 F" 0 "0.0 F" 0 "0.8 V" 0 "0.5" 0 "0.5" 0 "0.0" 0 "0.33" 0 "0.0 ps" 0 "0.0" 0 "0.0" 0 "1" 0 "600.0" 0 "0.0" 0 "1" 0 "1" 0 "0.0" 0 "0.0" 0 "0.0" 0 "0.0" 0 "0.0 m" 0 "0.0 m" 0 "0.0" 0 "1.0" 0 "1.0" 0 "26.85" 0 "26.85" 0>
  <MOSFET T1 1 560 340 34 -26 0 0 "nfet" 0 "300e-3" 1 "2e-5" 1 "0.0" 0 "0.6 V" 0 "0.0" 1 "0.0 Ohm" 0 "0.0 Ohm" 0 "0.0 Ohm" 0 "1e-14 A" 0 "1.0" 0 "1 um" 0 "1 um" 0 "0.0" 0 "0.1 um" 0 "0.0" 0 "0.0" 0 "0.0" 0 "0.0 F" 0 "0.0 F" 0 "0.8 V" 0 "0.5" 0 "0.5" 0 "0.0" 0 "0.33" 0 "0.0 ps" 0 "0.0" 0 "0.0" 0 "1" 0 "600.0" 0 "0.0" 0 "1" 0 "1" 0 "0.0" 0 "0.0" 0 "0.0" 0 "0.0" 0 "0.0 m" 0 "0.0 m" 0 "0.0" 0 "1.0" 0 "1.0" 0 "26.85" 0 "26.85" 0>
  <.TR TR1 1 60 400 0 59 0 0 "lin" 1 "0" 1 "5ns" 1 "100" 0 "Trapezoidal" 0 "2" 0 "1 ns" 0 "1e-12" 0 "150" 0 "0.001" 0 "1 pA" 0 "1 uV" 0 "26.85" 0 "1e-3" 0 "1e-6" 0 "1" 0 "CroutLU" 0 "no" 0 "yes" 0 "0" 0>
  <Vrect V3 1 340 310 18 -26 0 1 "0.9" 1 "1ns" 1 "1ns" 1 "10ps" 0 "10ps" 0 "0 ns" 0>
</Components>
<Wires>
  <160 200 160 250 "" 0 0 0 "">
  <160 200 260 200 "vdd" 250 170 58 "">
  <40 200 40 250 "" 0 0 0 "">
  <40 200 140 200 "vss" 120 170 43 "">
  <440 420 560 420 "vss" 490 390 17 "">
  <560 160 560 190 "" 0 0 0 "">
  <440 160 560 160 "vdd" 520 130 43 "">
  <560 420 590 420 "" 0 0 0 "">
  <580 220 590 220 "" 0 0 0 "">
  <590 160 590 220 "" 0 0 0 "">
  <560 160 590 160 "" 0 0 0 "">
  <500 220 530 220 "" 0 0 0 "">
  <500 220 500 280 "" 0 0 0 "">
  <340 280 500 280 "vin" 420 250 50 "">
  <590 340 590 420 "" 0 0 0 "">
  <580 340 590 340 "" 0 0 0 "">
  <560 370 560 420 "" 0 0 0 "">
  <560 250 560 310 "vout" 690 280 43 "">
  <500 280 500 340 "" 0 0 0 "">
  <500 340 530 340 "" 0 0 0 "">
</Wires>
<Diagrams>
  <Rect 360 660 240 160 3 #c0c0c0 1 00 1 0 1e-09 5e-09 1 -1 1 1.65943 1 -1 1 1 315 0 225 "" "" "">
	<"vin.Vt" #0000ff 0 3 0 0 0>
	<"ngspice/tran.v(vout)" #ff0000 0 3 0 0 0>
	<"ngspice/tran.v(vin)" #ff00ff 0 3 0 0 0>
  </Rect>
</Diagrams>
<Paintings>
</Paintings>

closed 2022-03-20 by ra3xdh