ra3xdh/qucs_s #61: ngspice no data from simulator?

ardasevinc, 2021-03-08:
I created a very simple DC circuit and simulated it. But on the diagram window dataset is empty.
I don't understand what's wrong.

[image]

ra3xdh, 2021-03-08:
DC simulation has no effect by itself if the Ngspice is selected as default kernel. Add AC or TRAN simulation and you will get the plots. Also try to press F8 or Simulation->Calculate DC bias and ypu will see the operation point of the circuit.

ardasevinc, 2021-03-09:
> DC simulation has no effect by itself if the Ngspice is selected as default kernel. Add AC or TRAN simulation and you will get the plots. Also try to press F8 or Simulation->Calculate DC bias and you will see the operation point of the circuit.

Thanks, I also combed through the docs and realized DC sim is not implemented as you said. Sorry for this dumb issue :d

closed 2021-03-09 by ardasevinc