ra3xdh/qucs_s #62: The following signatures were invalid: EXPKEYSIG 184B98BE3E68CD11 home:ra3xdh OBS Project

dreamcat4, 2021-04-11:
```sh
Err:6 https://download.opensuse.org/repositories/home:/ra3xdh/xUbuntu_18.04  InRelease                   
  The following signatures were invalid: EXPKEYSIG 184B98BE3E68CD11 home:ra3xdh OBS Project <home:ra3xdh@build.opensuse.org>

...

W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://download.opensuse.org/repositories/home:/ra3xdh/xUbuntu_18.04  InRelease: The following signatures were invalid: EXPKEYSIG 184B98BE3E68CD11 home:ra3xdh OBS Project <home:ra3xdh@build.opensuse.org>
W: Failed to fetch https://download.opensuse.org/repositories/home:/ra3xdh/xUbuntu_18.04/InRelease  The following signatures were invalid: EXPKEYSIG 184B98BE3E68CD11 home:ra3xdh OBS Project <home:ra3xdh@build.opensuse.org>
W: Some index files failed to download. They have been ignored, or old ones used instead.
$ 
```

pebbe, 2021-05-10:
Any chance for an updated signature?

ra3xdh, 2022-02-20:
The release key has been updated after the generation of new packages. The packages for Ubuntu 20.04 could be found here: https://download.opensuse.org/repositories/home:/ra3xdh/xUbuntu_20.04/

closed 2022-02-20 by ra3xdh