ra3xdh/qucs_s #63: TRIAC and other non linear components missing

ombra32, 2021-05-29:
Hi,
TRIAC and other non linear components (diac, thyristor, ecc..) are missing in Qucs-s when using Ngspice simulator.
Are presents using Qucsator simulator instead.

Have I missed something?
I'm using Qucs-s 0.0.22 appimage on Ubuntu 20.04.

[image]
[image]

ra3xdh, 2021-05-29:
Triac and thyristor are SPICE-incompatible. These are unique qucsator devices. That's why these devices are missing when SPICE engine is selected. Probably you may find the SPICE model for the specific thyristor device and attach it using the "SPICE file" component.

ombra32, 2021-05-29:
> Triac and thyristor are SPICE-incompatible. These are unique qucsator devices. That's why these devices are missing when SPICE engine is selected. Probably you may find the SPICE model for the specific thyristor device and attach it using the "SPICE file" component.

ok, I'll try..many thanks

closed 2022-02-05 by ra3xdh