ra3xdh/qucs_s #66: GUI on XFCE Linux Desktop cannot have bright background color on left pane

abdulbadii, 2021-07-11:
How come QUCS GUI on XFCE Linux Desktop cannot have bright background color on left pane so being not contrast ?
it'd make a user almost completely unable to see the object on it while screenshot on the manual PDF prove it is very good, visible
https://qucs.github.io/docs/tutorial/getstarted.pdf
https://qucs-help.readthedocs.io/_/down … .0.18/pdf/

it's so vexing
[image]
Please help

jorgemrisco, 2021-09-04:
I am having the same problema on KDE Plasma using the default "Breeze Dark" theme (that works perfectly on every other application)

It'd be nice to have an option to "toggle" a Default Color scheme tested for Qucs-s as a temporary fix

ra3xdh, 2022-02-14:
Yes, I know about this problem. There may be two solutions:

* Replace transparent background by white for all components icons like for SPICE devices. The icons will be visible for every theme, but it will look ugly;
* Recognize system theme and convert icons to inverse colors dynamically;

ra3xdh, 2022-02-22:
I have added a special icon set for dark themes. Now Qucs-S is able recognize dark theme automatically and apply the inverse icons from `bitmaps/dark` subdirectory instead of normal. For light theme old icons are used. This screenshot illustrates main window using KDE dark theme. The icon set was redesigned by Qucs-S user from https://linux.org.ru forum. This improvement is availabel at this branch https://github.com/ra3xdh/qucs_s/tree/dark I am also considering to replace old raster icons taken from KDE3 by modern SVG icons set to improve the application look on HiDPI displays.

[image]

ra3xdh, 2022-06-30:
The dark/light theme setting will be added in the upcoming 0.0.24 release.

closed 2022-06-30 by ra3xdh