ra3xdh/qucs_s #73: Fourier analysis with Xyce backend is broken

ra3xdh, 2022-02-16:
Fourier analysis with `.FOUR`  gives no output if the latest Xyce v7.4 is used. The Qt4 Qucs-S version is affected too.

tvrusso, 2022-02-16:
This is probably due to a change to .FOUR output that was put in place in Xyce 7.4.  According to the release notes:

[image]

tvrusso, 2022-02-17:
So instead of looking in "netlist.four" you have to look in "netlist.four0" even if it's not a .step netlist.  If the netlist *DOES* have .step in it, then each step's fourier output is in a different file.

ra3xdh, 2022-02-17:
Thank you for pointing this. I will correct the Fourier analysis support in the next few days.

ra3xdh, 2022-02-18:
I have fixed a `.FOUR` simulation for Xyce. Old circuits are operational again now. The combination of `.FOUR`+`.STEP` will require a bit more work, because it will be needed to modify the simulation output parser.

[image]

ra3xdh, 2025-06-01:
Closing as stalled issue. Partly fixed.

closed 2025-06-01 by ra3xdh