ra3xdh/qucs_s #78: qucs-s 0.23 compiles, runs but no data results display

tckosvic, 2022-03-02:
Compiled qucs-s 0.23 and ran simple tutorial #1 (voltage divider) after creating schematic.  Looked to run ok.  Simulation dialogue showed:

`Ngspice started...

No compatibility mode selected!

Circuit: * qucs 0.0.23 /home/tom/.qucs/tuts_prj/dc-1.sch

ngspice-36 done
Simulation finished`

Data page *.dpl is created.  Pasting a cartesian table into that page as per tutorial shows no results just blank fields.

The AppImage file for qucs-s 0.23 also runs problem but presents no results.

I have plain qucs 0.19 linux compiled on my machine.  That runs divider tutorial fine and shows results.
I also have windows qucs 0.19 in wine that also runs and presents results.

Any ideas as to cause of problem?

tom kosvic

ra3xdh, 2022-03-02:
Hello Tom,
it is not a bug in Qucs-S. The DC simulation has no effect with Ngspice. You need to add TRAN simulation to see the result. Look a the example at the `ngspice` subdirectory. The Qucs tutorial partially is not valid when the Ngspice is used as the default simulation kernel and need to be updated.

ra3xdh, 2022-03-06:
Probably it make sense to give a warning when only DC simulation found in the schematic. For Ngspice/Xyce it has no effect.

ra3xdh, 2022-03-09:
The list of actions to prevent such issues in the future:

- [ ] Update Chapter 1 of the documentation and provide proper tutorial; don't use tutorial from Qucs;
- [ ] Show a warning when only DC simulation is presented in the circuit; this simulation has no effect with Ngspice/Xyce;

ra3xdh, 2022-03-28:
The links in the Help menu are pointing now only to to the Qucs-S tutorials and help documents. The links to old Qucs help are removed because otherwise it can be misleading for new users.

closed 2022-03-28 by ra3xdh