ra3xdh/qucs_s #79: Add S-parameter  simulation support for Xyce

ra3xdh, 2022-03-06:
The recent version of Xyce support now the S-parameter simulation with `.LIN` directive. It's need to provide user an automated way for this simulation type. Currently the S-parameter simulation may be used only with XYCE-script simulation type. 

[image]

ra3xdh, 2022-03-12:
I have implemented basic support of the S-parameter analysis with Xyce on the `current` branch. I would ask all interested persons to test this new feature. The attached screenshot illustrates how it works.

Known issues
* Xyce STD parser cannot reassemble the Re() and Im() parts of the simulation variable and plots them independently.
* The Qucs functions stoy(), stoz() etc. will not work. The Xyce puts the Y and Z matrix to the output. 

[image]

MikeBrinson, 2022-03-13:
Hello Vadim,
Thanks for adding the S-parameter simulation icons to Qucs-S\Xyce.
I will add Y and Z etc. through a series of library symbol probes.In principle this is possible but will require some experimentation.
On a different topic I noticed that the sensitivity simulation features for Ngspice areout of step.  Ngspice allows dc and ac sensitivity, while Xyce allows dc and tran.Xyce ac sensitivity analysis was added after Qucs-S 0.0.22 was released.
AC sensitivity analysis can of course be run using a XYCE script but it keeps Ngspiceand Xyce in step if a simulation icon was added to complete Xyce sensitivity analysisfeatures.
I will keep you informed of my progress on Qucs-S tasks.
Best wishes,
Mike

Mike Brinson ***@***.***

 
 

   
   -  Add support for 50-ohm port devices
   -  Implement conversion of the S-parameter simulation in the .AC+.LIN netlist entry.
   -  Generate SPICE netlist for Qucs microwave devices
   -  Document S-parameter analysis with Xyce ( @MikeBrinson )

ra3xdh, 2022-03-13:
Hello Mike,
The Y and Z parameters are presented in the dataset by default. It's not to add them as the probes separately. Look at the attached screenshot. It's possible to select and plot S,Y, or Z parameters set. 

[image]

ra3xdh, 2022-03-13:
I have also implemented reassembling of the complex numbers in the Xyce STD output files. Now the S-parameter simulation results could be plotted on Smith and polar diagrams.
[image]

MikeBrinson, 2022-03-13:
Hello Vadim,
Thanks for your last emails.
I was not sure you had converted the simulated S-parameters to Y and Z.This makes using the Xyce S-parameter simulation more straightforward.
The next step will be to build a library for RF/microwave components.  This willtake time.  Unfortunately, the original microstrip components are for ac simulationonly.  Work needs to be done on builder more general models that work in thetransient domain as well as the ac domain. 
In the background I will give some thought as to how we progress with these models.
Best wishes.
Mike   

Mike Brinson ***@***.***

ra3xdh, 2022-03-13:
Hello Mike,

in general it will be needed to implement the `spice_netlist()` method for Qucs microwave devices. Then the Qucs schematic containing RF devices could be opened and simulated with Xyce.  But conversion of the Qucs microstrip devices into the SPICE netlist entries may not be a trivial task. There is no direct replacement for such devices in SPICE. It will be needed to somehow convert the Qucs devices into SPICE T and O(LTRA) devices or use equivalent circuits. 

But nevertheless I am inclined to include the S-parameter analysis with Xyce in the next release. At least it will allow to simulate the circuit containing semiconductor devices and lumped (RCL) components.

Regards,
Vadim

MikeBrinson, 2022-03-14:
Hello Vadim,
Thanks for your note on Qucs-S RF models.
In the past I worked in RF/microwave research.Hence my knowledge of modeling RF components withSPICE is up-to-speed.
The addition of S-parameter simulation to Qucs-S/Xyce is a veryimportant milestone.  Once S-parameter simulation is releasedfor Ngspice Qucs-S will become a very important tool in theRF/microwave domains.  
My proposal for RF modeling etc is as follows:
To build a RF/Microwave component library of the common components, forexample passive components, lines, couplers, power dividers and microstripelements.  As you point out these will in general be based on the SPICE T, O(LTRA)and electrical equivalent circuits.  Many of these models I have built and tested in the past.  In the long term I will attempt to develop Verilog-A models for the RF models.This should improve overall performance.  Conversion  to Verilog-A will however, largely dependon future availability of Verilog-A on the fly compilation facilities to Qucs.S/Ngspice/Xyce.
Adding the Xyce S-parameter simulation to Qucs-S 0.0.24 will help in this work.
I can see from present progress the next few months are going to full of interesting tasks.At present I am concentrating on writing the new Qucs-S documentation, testing and willadd RF/microwave modeling to my list.
Best wishes.
Mike

Mike Brinson ***@***.***

ra3xdh, 2022-07-01:
The S-parameter simulation with Xyce is available in the 0.0.24 release https://github.com/ra3xdh/qucs_s/releases/tag/0.0.24

closed 2022-07-01 by ra3xdh