ra3xdh/qucs_s #80: Qucs and Qucs-S use the same application setting configuration settings

p-brane, 2022-03-09:
Qucs-S and Qucs use the same File -> Application Settings.. -> Location configuration settings/file. Where are the File -> Application Settings.. configuration stored? How do you set Qucs to use a specific File -> Application Settings.. -> Locations configuration/file? I would like Qucs-S to use a different configuration from Qucs so they operate independently of each other.

tvrusso, 2022-03-09:
Vadim fixed this a bit over two weeks ago in #71 

Qucs-S and Qucs no longer use the same settings file as of release 0.0.23.

ra3xdh, 2022-03-09:
@p-brane What version are you using? The settings file since is located for Linux in `$HOME/.config/qucs/qucs_s.conf` and for Windows the settings are stored somewhere in the registry. Qucs and Qucs-S still share the same `$HOME/.qucs` location for projects and libraries but I decided to keep it for backward compatibility.

ra3xdh, 2022-03-20:
No further explanations provided. Closing.

closed 2022-03-20 by ra3xdh

p-brane, 2022-03-21:
Hi @ra3xdh ,
I was using version 0.19. I've download version 0.23 and will report back when I've had a chance to test the new version.