ra3xdh/qucs_s #83: Insert Equation

jleitch2013, 2022-03-14:
Qucs- 0.0.23:

Using the Transient simulation and also inserting equations.

The simulation runs, however, the equation simulation does not seem to be sending the results of the equations to the data output for display.

ra3xdh, 2022-03-14:
Please provide more details. I have tested equations before the releasing and didn't found problems. Also the equation may have the syntax incompatible with SPICE. What backend are using? If you are using Xyce the equations will not work, because the Xyce has no postprocessor.

ianloic, 2022-03-18:
When I run the [ngspice/Distortion.sch](https://github.com/ra3xdh/qucs_s/blob/master/examples/ngspice/Distortion.sch) example with ngspice as the simulator the first diagram isn't displaying anything:
[image]

Also, when I go to add a new diagram the value `K` defined in the equation `Eqn1` isn't available:
[image]

ianloic, 2022-03-18:
Okay, and now playing around and reading the ngspice output I'm realizing that I was just using this wrong.

ra3xdh, 2022-03-18:
It's becoming clearer now. It is not a bug. The gain calculation will not work with the distortion simulation. The equation in the example may mislead user and need to be corrected.

ra3xdh, 2022-03-20:
Distortion example is fixed on the current branch by https://github.com/ra3xdh/qucs_s/commit/0b30f80dc84a1ee4dc2bb82dcd26ec0060c07bca

closed 2022-03-20 by ra3xdh