ra3xdh/qucs_s #84: Add S-parameter support for Ngspice

ra3xdh, 2022-03-20:
Related to #79 S-parameter simulation `.SP` will be added in Ngspice-37. Currently it is available from Ngspice Git.  It's need to implement it in the Qucs-S. The following tasks need to be done:

- [ ] Provide netlist entries for 50-Ohm ports and S-parameter simulation
- [ ] Provide equivalent for Qucs stoz() and stoy() functions, because currently Ngspice can output only S-matrix
- [ ] Provide netlist entries for other microwave devices 
- [ ] Document new feature

JontAlle, 2022-03-26:
HI to all S-parameter users.
This is a topic I have some experience with, and I would like to help in this development. 
Please look at my book (online pdf), at 
   http://jontalle.web.engr.illinois.edu/MISC/Springer.19/2020_Book_AnInvitationToMathematicalPhys.pdf
You will find several discussions relevant to the S-parameters, along with Impedance and transmission matrices.
For example search in the PDF for "reflectance" (e.g., p. 190, 215, 259, ...) or "transmission matrix" a.k.a. "ABCD," or look at page 150 for a discussion of how to use the impedance matrix.
Specific comment:
Perhaps a useful contribution would be to add pole-zero plots of the S-parameters. Is that presently being done?  I'll look at the Nspice users manual to see if that is presently being done.

Prof Jont Allen (jontallen@ieee.org, auditorymodels.org)

holvo, 2022-03-29:
There is an update to RF for ngspice available at a new ngspice git branch s-parameters-2.
It includes  Y- and Z-matrix outputs and noise simulation with C-matrix output.

An update to the manual is still needed.

Do you have some circuit examples for me to add to the ngspice examples distribution?

Holger

holvo, 2022-03-29:
I have just made an update to the ngspice manual (chapters 4.1.11, 15.3.8, and 17.5.80), see http://ngspice.sourceforge.net/docs.html.

ra3xdh, 2022-03-29:
Hello Holger,
nice to hear news from the Nsgpice team. I will test the new features added in the `s-parameters-2` branch. It should resolve the problem with stoy() and stoz() functions equivalent. I am using the wideband HF-amplifier from the book E.Red "HF-Module in 50-Ohm-Technik" for testing of the S-parameters simulation. Here I attach a Qucs-S schematic (gzipped *.sch file) using XYCE S-parameter simulation. The netlist could be reused as the test example for the Ngspice after some manual adjustments. The auto-generated netlist is here: https://gist.github.com/ra3xdh/b9c7e68897e35cd2e33c3e8a151b0f85

[Red_amp_xyce.sch.gz](https://github.com/ra3xdh/qucs_s/files/8371751/Red_amp_xyce.sch.gz)

[image]

ra3xdh, 2022-04-02:
I have added a basic support for S-parameter simulation for Ngspice+Qucs-S on the `current` branch. The Ngspice should be compiled from git from `s-parameters-2` branch. I will implement support for S-parameter noise simulation later. The attached screenshot illustrates the simulation of the wideband BJT amplifier for HF transceiver. As you can see, the results from Ngspice and Xyce are identical.
[image]

ra3xdh, 2022-04-03:
I have just added support for Ngspice S-parameter and Noise analysis using `donoise` parameter. This feature is available on the `current` branch.  The attached screenshot illustrates its operation. 
[image]

ra3xdh, 2022-04-03:
Examples added by https://github.com/ra3xdh/qucs_s/commit/c4c625236925e32ca0e704c201e6e72aab23cf53

tomhajjar, 2022-04-05:
The latest released Windows version of Qucs-S 0.0.23 does not have the "pac" source. When will this be available?

[image]

ra3xdh, 2022-04-05:
Hello Tom,
the S-parameter simulation is expected for 0.0.24 version. I am planning to prepare the packages and make a release in the next few weeks. Currently you may get access to this simulation type only if you compile Qucs-S from source. You may need to build the latest Ngspice36+ from source too.

tomhajjar, 2022-06-09:
Any update on the 0.0.24 version?

ra3xdh, 2022-06-09:
Hello Tom,

currently two tasks are remaining for 0.0.24:
* New FFT analysis #88 
* Bug with diode Ikf parameter for Ngspice-37 https://sourceforge.net/p/ngspice/mailman/message/37647791/ 

The Ngspice version from the latest git cannot process the default Qucs `Ikf=0` value. The Xyce doesn't complain about it. If no solution will be provided by Ngspice team I need to add a fix into the Ngspice netlister. This bug is currently preventing me from the releasing of 0.0.24.

tomhajjar, 2022-06-14:
Vadim

Will the "RF" components in Qucs 0.0.20 be enabled in the next version of Qucs-S? Most seem to be in the source for Qucs-S.

Transmission lines like microstrip and coplanar elements?

RF Components including the "new" Inductor and Capacitor with "Q"?

ra3xdh, 2022-06-14:
Hello Tom,
the microstrip and transmission line are unique components introduced by Qucsator. These devices have no equivalent in SPICE. It's need to somehow provide SPICE netlist entries for these devices using the  `T`, `O` and `U` SPICE devices. I didn' started this task yet, so these devices will be not presented in the upcoming Qucs-S version. But it's possible to add the LC-devices with Q using the equivalent circuits. The CAPQ and INDQ devices will be added in the 0.0.25 version.

ra3xdh, 2022-07-01:
The S-parameter simulation with Ngspice>=37 become available in the 0.0.24 release (published on July, 01). The simulation of the active and passive circuits containing lumped device is currently implemented. The implementation of the microstrip devices model will be considered as the separate task for the next releases.

tomhajjar, 2022-07-01:
Vadim

I tried to display S-parameters in dB like in Qucs but no luck. Since there are no errors I cannot figure out what's wrong. 

[image]
[schematics-S.zip](https://github.com/ra3xdh/qucs_s/files/9030849/schematics-S.zip)

ra3xdh, 2022-07-01:
@tomhajjar You need to use Nutmeg equations with Ngspice (Spice specific sections->Nutmeg equation). The Qucs equations may have no effect and kept for backward compatibility. Also you need to select a simulation related to this equation in its properties. Since 0.0.24 the Qucs-S will insert Nutmeg equation on schematic by default if the Ngspice is selected as the simulation engine.

tomhajjar, 2022-07-01:
Does the Group Delay function in ngspice work in this case? GD=group-delay(s_2_1) crashes npspice. 

Where is the "log" for ngspice? When ngspice crashes, the message lashes so fast it makes it hard to identify the problem(s).

[Qucs-S S-parameters_ngspice_prj.zip](https://github.com/ra3xdh/qucs_s/files/9031954/Qucs-S.S-parameters_ngspice_prj.zip)
[image]

tomhajjar, 2022-07-01:
I got it working. 

The group_delay function in ngspice has issues. The plot has large discontinuities in-band as seen in the 3plpf circuit. The group delay function in Qucs 0.0.20 works better. 

[Qucs-S S-parameters_ngspice_prj.zip](https://github.com/ra3xdh/qucs_s/files/9031984/Qucs-S.S-parameters_ngspice_prj.zip)

ra3xdh, 2022-07-02:
@tomhajjar Yes, it seems the group_delay function in Ngspice has the wrong implementation. A bug report should be submitted to Ngspice mailing lists.

tomhajjar, 2022-07-02:
Holger Vogt over at the sourceforge ngspice discussion came up with a workaround. ngspice will be fixed in a future release. 

Group_Delay = -1*deriv(cph(S_2_1))/2/pi

closed 2022-07-11 by ra3xdh