ra3xdh/qucs_s #86: Ngspice error

bladane, 2022-04-11:
Installed qucs-s 0.0.23 for ubuntu20.04 (followed procedure from the site: [https://ra3xdh.github.io/#docs] ). 

It also installed ngspice-31 that I checked with simple *.cir example and it is working. So Ngspice is working.

Then I opened Qucs 0.0.23 and created an example of RC schematic from tutorial (prior to this set ngspice as default simulator and directed to Ngspice executable location: /usr/bin/ngspice) but when tried to simulate it is showing:
- popup window **Simulate with external simulator** with message in Simulation console - **Ngspice started**  and 
- new popup window with message **Failed to start simulator**. 
- And  when **Failed to start simulator** popup is closed it writes inside Simulation console new message - **Ngspice error...**

Any recommendation how to debug this problem.

ra3xdh, 2022-04-11:
This may be related to #85 Please send me your `qucs-s.conf` file. I cannot reproduce this issue on my clean Ubunutu20.04 installation. It seems that the path for netlist file somehow set is unwriteable.

bladane, 2022-04-11:
here is the content of qucs-s.conf (renamed to txt):
[qucs_s.conf.txt](https://github.com/ra3xdh/qucs_s/files/8463264/qucs_s.conf.txt)
Thnx!

ra3xdh, 2022-04-11:
Yes, my suspicions were correct. The working directory is not writeable. Please also send me screenshot from the `Simulation->Select default simulator menu`. I am interested what is this setting:
[image]

bladane, 2022-04-11:
Here is the screenshot - 

[image]

bladane, 2022-04-11:
Fixed it by changing the directory to my home dir. Now it is working!
Thank you one again!

ra3xdh, 2022-04-11:
As a workaround you may to set "Directory to store netlist ..." as the writeable path. Usually it should be `$HOME/.qucs/spice4qucs/`. I will check why it may be not set correctly automatically.

bladane, 2022-04-11:
Done as per your suggestion.

ra3xdh, 2022-04-11:
Yes, I can reproduce an issue now. The Qucs-S may set the working directory as an unwriteable path. It's strange that it was not discovered before. I will provide a fix soon.

ra3xdh, 2022-04-11:
Fixed by https://github.com/ra3xdh/qucs_s/commit/1fbe2c74123b46d22d33c2708deeef5aa7dd9787 on `master` branch. The fix will be available in 0.0.24 release.

closed 2022-04-11 by ra3xdh