ra3xdh/qucs_s #88: Add spectrum analysis simulation for Nsgpice

ra3xdh, 2022-05-09:
Currently the FFT analysis is possible using the Nutmeg scripting https://qucs-s-help.readthedocs.io/en/latest/ASim.html#spectrum-analysis-with-ngspice-and-nutmeg-scripting It would be more user-friendly way if we add a special simulation type that generates this Nutmeg script automatically.

ra3xdh, 2022-05-12:
The attached screenshot illustrates the concept of the new FFT simulation. This simulation type is available on the `current` branch.
[image]

closed 2022-06-29 by ra3xdh

tomhajjar, 2022-08-04:
I have never gotten the example in Figure 5.8 to work using the following Nutmeg script. No data is generated. 

tran 1n 10u 0
linearize v(out)
fft V(out)
let S = db(v(out))

This new method will save time as well. 
[image]

ra3xdh, 2022-08-05:
Hello Tom, the schematic from the manual works on my machine as expected. Probably you have something wrong with the Nutmeg script setting. Please check the following attached schematic: 
[Mixer.zip](https://github.com/ra3xdh/qucs_s/files/9267342/Mixer.zip)

tomhajjar, 2022-08-05:
When I copy your Nutmeg script and tran statement into my schematic, it still didn't work. The only difference I found was my schematic name has spaces in it. Changing name to remove spaces fixed it. Not sure why this happens. I have many schematics with names using spaces. 

Changed:
Double Balanced Diode Mixer 7-15MHz.sch

To:
DBM_Mixer.sch

Your schematic has two different tran statements. I assume using Nutmeg script this is allowed?

ra3xdh, 2022-08-05:
Yes, I can reproduce this bug now. The space in the schematic name breaks the custom simulation and produce the string like this in the netlist:
~~~
write Mixer 1_custom.txt V(out) S
~~~
The `write` command become invalid. I will fix this bug in the next release. As workaround you may save your schematic using scripting simulation without the space in the file name. Other simulation types are not affected.

ra3xdh, 2022-08-05:
I have created a separate issue #129 instead of writing to the closed issue.