ra3xdh/qucs_s #89: Qucs_s with ngspice not working correctly

kwmartin, 2022-05-19:
I have verified that with a hand modified netlist/control file that ngspice seems to be working correctly, but with the getting started example, with a dc simulation and a tran simulation,  the netlist does not seem to be made correctly, the display blocks (tabular and timing diagram) are not being filled in. Here is the netlist generated:

* Qucs 0.0.23 /home/martin/.qucs/firstProject_prj/schem1.sch
.INCLUDE "/usr/local/lib/share/qucs-s/xspice_cmlib/include/ngspice_mathfunc.inc"
* Qucs 0.0.23  /home/martin/.qucs/firstProject_prj/schem1.sch
R2 divide Vdd  50
R1 0 divide  100
V1 Vdd 0 DC 0 PULSE(0 1 0 1N 1N 0.000999998 999998) AC 0
.control
echo "" > spice4qucs.cir.noise
echo "" > spice4qucs.cir.pz
tran 9.09091e-05 0.001 0 
write schem1_tran.txt v(Vdd) v(divide) 
destroy all
reset

exit
.endc
.END

kwmartin, 2022-05-19:
Add on, when doing a biasing point simulation, I get the error pop up stating:
No Ground found. Please add at least one ground!
Simulation finished
Ngspice error...

ra3xdh, 2022-05-20:
Hello Kenneth,

What tutorial are you using? The Qucs tutorial is not valid for Qucs-S. The Ngspice contains many incompatible features with old Qucs. Digital simulation will not work, beacause digital components are not compatible with Ngspice. I have added a new tutorial explaining simulation with Ngspice: https://ra3xdh.github.io/pdf/qucs_s_tutorial.pdf See also #78 In the 0.0.24 the link for Qucs tutorial will be remove from the Help menu and replaced by Qucs-S tutorial link.

The timing diagram is valid only for digital simulation and is Ngspice-incompatible. The diagrams should not be presented in the netlist and are saved only in schematic `*.sch` or `*.dpl` files despite the diagrams are the components on schematic.

The "No ground" message is not a bug, but fundamental Ngspice limitation. If the circuit has no ground node, it cannot be simulated. Insert at least one ground.

kwmartin, 2022-05-26:
Vadim, the verilog netlisting does not seem to be in qucs-s; is this 
something that might re-appear soon, or not for awhile? Also, are there 
any plans to add buses to the netlisting? I've been working my way 
through the code to try and understand the flow, without a roadmap, it's 
not an easy task (for me anyways). Thanks.

-Ken

-- 
Kenneth Martin
Mobile: +1(416) 897-1940
www.granitesemi.com

ra3xdh, 2022-05-26:
The Verilog devices are not supported by Ngspice and therefore the Qucs-S has no digital Verilog netlister. In the future I am planning to implement disabled Qucs digital components using Ngspice XSPICE devices or LTspice-compatible devices that will be added by Ngspice team in the furture. But currently the implementation of the digital simulation is not a high priority task for me.

ra3xdh, 2022-05-26:
The buses make sense mainly for digital circuits. At first it will be needed to make the digital components operational with Ngspice in a some way and then to add buses.

kwmartin, 2022-05-26:
Vadim, thank you for getting back to me. I might also mention that 
qucs-s in Ubuntu 20.04 is messing up Qt-5 based windows such as 
KDE-based Dolphin file manager (turned the scheme yellow). I had to use 
qtsct to get the "simple" scheme back working; re-booting alone was not 
enough to fix.

-Ken

-------- Forwarded Message --------
Subject: 	Re: [ra3xdh/qucs_s] Qucs_s with ngspice not working correctly 
(Issue #89)
Date: 	Thu, 26 May 2022 11:56:57 -0700

<https://github.com/ra3xdh/qucs_s/issues/89#issuecomment-1138907788>, or 
unsubscribe

holvo, 2022-07-02:
We have started development for digital devices in ngspice. ngspice will support PSPICE U devices, which are basic building blocks for digital devices of the 4000 and 7400 series. These U devices are translated to XSPICE digital devices. A first commit has been made to ngspice git branch pre-master. Still missing (but under development as well) is an automatic linking between digital event nodes and analog nodes. Goal is to have this readily available for ngspice-38.

ra3xdh, 2022-07-03:
Hello Holger, I have seen the dedicated commits on `pre-master` branch of the Ngspice Git. The implementation of the U-device will allow to bring the digital simulation back to Qucs-S. I have created a special issue for this task. See #97

ra3xdh, 2022-07-09:
No further news from issue reporter since 1.5 months. I assume the problem is resolved. Closing.

closed 2022-07-09 by ra3xdh