ra3xdh/qucs_s #92: Qucs-S GUI improvements

ra3xdh, 2022-06-09:
I have recently received some feedback from Qucs-S users and the following GUI improvements were proposed:

- [ ] Implement exact marker position setting on plots. Currently the marker position could be exactly set only with arrow button. It would be good to add some input field in the marker properties dialog to set an exact X-axis position;
- [ ] Enable show of the average and RMS values on schematic like for DC operation point; 
- [ ] Add wattmeter probe like voltmeter and ammeter;
- [ ] Add dB output for the Y-axis to avoid using of the equations to get the dB output.

tomhajjar, 2022-06-10:
It would be nice to be able to set the format of the X scale. I've tried the trick below but it has issues. This was an old request for Qucs.

https://sourceforge.net/p/qucs/mailman/message/35975391/

ra3xdh, 2022-06-11:
@tomhajjar Yes, I will consider a way how to add units like 1k, 1M, etc. for X-axis. The slight modification of the diagram properties will be required.

ra3xdh, 2022-06-13:
I have just implemented the numeric marker position setting. The screenshot illustrates how it looks like. This feature is available on the ´current´ branch now. 
[image]

ra3xdh, 2022-07-03:
I have added a switch to enable engineering notation (1k, 1M, 1G etc.) for numbers on diagrams. Attached screenshot illustrates how it looks now. This feature is available on `current` branch.
[image]
[image]

ra3xdh, 2022-07-12:
I have just added a possibility to set dB units for Y-axis if logarithmic scale is selected. The new feature is available on `current` branch.
[image]
[image]

ra3xdh, 2022-08-08:
I have noticed the following problems with dB units on Y axis:
* The marker values still be in absolute units while the axis is set as dB
* It is impossible to set limits and grid step because limits still be in absolute units
* Grid step setting is locked for log axis but should be unlocked for dB log axis

tomhajjar, 2022-08-15:
Qucs/Qucs-S are lacking when one tries to choose items in the library or you want info on a placed device. There is zero info on the devices unless you use the Library Tools. Try to pick a varistor from the lengthy list.

It would be nice if the Main dock showed a components "Info" after a mouse click.

Show a placed devices "Info" via a new choice after a mouse right button click. Edit properties shows little info.

[image]
[image]
[image]

ra3xdh, 2022-08-16:
Yes, the library tree on the left panel may benefit from suggested improvement. I was also considering such addition. I will implement symbol and device info preview in one of the next releases.

ra3xdh, 2022-08-16:
I have created a separated task #135 for library tree improvement.

ra3xdh, 2022-08-24:
I have fixed setting limits and marker issue when the dB units are selected for axis. The remaining issues is the manual step setting for logarithmic dB axis.

ra3xdh, 2022-09-18:
I have noticed application crash when adding a curve to Smith diagram. This is related to newly introduced dB axis units.

dsm, 2022-11-23:
I tested app on windows 10 same issue here

ra3xdh, 2022-11-23:
@dsm , Yes, I can reproduce this bug. I will provide a fix. I have crated a special dedicated issue #187

dsm, 2022-11-25:
Let's make the necessary arrangements to compile the application with qt6 to support and maintain it for a longer period of time. I know priority lower than usability, add new feature and make it more robust app and I know this needs so many code clean up maybe rewrite so many code but we need well clean code base for good maintainability for future release

ra3xdh, 2022-11-25:
I didn't try to compile Qucs-S on Qt6 yet, but the adding support of Qt6 should be much more smooth that Qt3->Qt4 migration. Qt6 doesn't contain changes breaking backward compatibility. If the application compiles on Qt5.15 without deprecation warnings, it should be compiled on Qt6 too. I have done some works in this direction like removing deprecated QtScript dependency. See the #75 and #77.

ra3xdh, 2022-11-25:
I have created a special issue dedicated Qt6 portability. See #190

ponyatov, 2023-04-17:
It would be great to have rich scripting facilities (Python?) to making more or less complex things in the running QUCS instance.

At least, I want to use QUCS as a schematic frontend via bidirectional data interchange with PCB making software such as KiCAD, Specctra and TopoR.

ra3xdh, 2023-04-20:
@ponyatov Yes, the adding of the Python scripting features were discussed for old Qucs around six years ago. Currently we only provide a Python package to load Qucs dataset and plot the diagrams. Something like Python extensions in Klayout may be useful. But it is totally unclear how to proceed with adding Python. Currently I have more high-priority tasks likes implementing of the microstrip devices and reverting back the digital simulation. So, don't expect adding Python integration in the near future.

eldaravt, 2023-06-19:
> I have just added a possibility to set dB units for Y-axis if logarithmic scale is selected. The new feature is available on `current` branch. [image] [image]

[image]
Incorrect:
- left view use log. left axis (-6dB at cutoff freq.)
- right view Nutmega eq (-3dB at cutoff freq.)

ra3xdh, 2023-06-19:
@eldaravt I have just added a fix for this. 
[image]

johndoespiced, 2023-10-12:
Sorry if this is the wrong place. But it would be neat if Qucs-S could have as solver engines the ones only available through Wine, like QucsStudio, QSpice and LTSpice. I think there are command line executables for these.

ra3xdh, 2023-10-12:
> Qucs-S could have as solver engines the ones only available through Wine, like QucsStudio, QSpice and LTSpice

Adding support for Windows-only proprietary simulation backends is not considered. Qucs-S supports only open-source cross-platfrom simulation kernels.

SmallJoker, 2023-12-10:
> User request: automatically align device by grid after rotation

I have run into the same issue after rotating a generic LED or transistor. Clicking "Snap to grid" did fix the alignment issue for the selected component. Is there any particular reason why this alignment function cannot be run after rotating the part?

PS: Thank you for continuing the Qucs project. Much appreciated.

tomhajjar, 2023-12-10:
In #203 I asked if Rotate "Right" could be added.

I noticed that using Rotate "Left" multiple times caused the selected items to jump off the grid. Adding Rotate "Right" would mean both less mouse clicks and minimize grid misalignment.

ra3xdh, 2023-12-10:
> Is there any particular reason why this alignment function cannot be run after rotating the part

I cannot say why the developers of the schematic widget didn't add align to grid after rotation. I have tried to fix this issue recently, and this was not a straightforward task. Need to dig into the legacy code. I will try to provide a fix for this by the next release.

ra3xdh, 2023-12-12:
I have just added a fix for device rotation. Now the device should be aligned on grid if it was rotated. This fix is available on `current` branch. I cannot reproduce the rotation issues anymore after the patch has been applied. I would kindly ask to test it before release if the issue still present.

ra3xdh, 2023-12-16:
The recent attempt to add an automatic set on grid after rotation made the things worse. Now the device always falls out of grid after the copy/paste from clipboard. I will retract the recent commit if the solution will not be found.

tomhajjar, 2023-12-16:
Adding Rotate "Right" would mean less mouse clicks and minimize grid misalignment... ;-}

ra3xdh, 2023-12-16:
@tomhajjar The adding "Rotate right" may be harder than fix the rotation. It's need to dig into the legacy code to implement this. I cannot exactly say what is the reason of grid misalignment. This bug is hard to reproduce and appears time to time. I cannot determine exact steps to reproduce it. It appears on Windows more often than on Linux. It seems it depends on the platform and screen resolution.

tomhajjar, 2023-12-16:
If you rotate the tunnel diode symbol 3 times, the 45 deg lines jump off grid.

[image]

[symbols_rotate.zip](https://github.com/ra3xdh/qucs_s/files/13692832/symbols_rotate.zip)

ra3xdh, 2023-12-16:
> If you rotate the tunnel diode symbol 3 times, the 45 deg lines jump off grid

This is not related to device rotation bug. The painting rotation works differently. And it cannot be fixed by adding rotate right function. The painting rotation was fixed by #397 , but the device grid misalignment still appear time to time.

iwbnwif, 2024-01-20:
> I have just implemented the numeric marker position setting. The screenshot illustrates how it looks like. This feature is available on the ´current´ branch now. [image]

I think it is relatively simple to add a small square or triangle to the end of the marker leader line. I have a test version working and it is helpful, especially with the very awesome new tuning function. Would it be preferable to include an option in the marker dialog for example "Marker indicator:" "Off" / "Square" / "Triangle"?
I am happy to provide a PR if this is of interest.

ra3xdh, 2024-01-20:
@iwbnwif , Yes, please open a PR. This will make marker position more visible. One marker indicator type would be sufficient.

iwbnwif, 2024-01-20:
> @iwbnwif , Yes, please open a PR. This will make marker position more visible. One marker indicator type would be sufficient.
I have submitted a PR which has both square and triangle as options. The reason is that I find the square more useful on return loss graphs, but the triangle more useful on most other graphs.

tomhajjar, 2024-02-02:
Qucs can export a Touchstone formatted s-parameter file using qucsconv.exe. This can be useful for some users. 

qucsconv converts the s-parameter data in filename.dat. Oddly qucsconv defaults to naming it to filename.dat instead of filename.s2p. 

[image]

ra3xdh, 2024-02-03:
> Qucs can export a Touchstone formatted s-parameter file using qucsconv.exe

This feature will be returned after implementing #370 Qucsconv has a deep dependency on Qucsator and cannot be divided from Qucsator. The full qucsconv will be returned after adding `qucsator_rf` as a submodule.

dsm, 2024-04-12:
maybe switch qt 6.7 or future lts 6.8 default for windows it's looks nice on win11 with new windows11 style on dark theme.

[image]

ra3xdh, 2024-06-09:
This issue still be opened since 2022 and I decided to close this with the following resolution on remaining tasks:

* Adding wattmeter is not straightforward task. The experiments with the SPICE netlsit has showed that the power calculation may be not reliable for all analyses cases. Some analyses require adding multiplier on schrmatic, but something works only using postprocessing scripts. Let's postpone this task for now.
* Automatic enumeration of the devices may be implemented in the future. I will open another ticket when this task will be planned for the next release.

closed 2024-06-09 by ra3xdh