ra3xdh/qucs_s #93: Qucs-S uses wrong symbol for enhancement mode MOSFETS

tomhajjar, 2022-06-10:
Qucs-S and Qucs is using the symbol for depletion mode MOSFET when placing an enhancement mode MOSFET onto the schematic. The symbol is correct in the left component or library panel and is correct when dragging the component onto the schematic BUT when you click the mouse the wrong symbol is placed. Doesn't matter if it is the n, p or version with substrate. It always places the depletion version of the device.

[image]

tomhajjar, 2022-06-10:
Sorry I put Qucs first. Attached is Qucs-S doing same thing. 

[image]
g

ra3xdh, 2022-06-11:
Yes, the MOSFET library was inherited from Qucs and the behavior will be the same. It seems that the default symbol in the library is wrong.

ra3xdh, 2022-06-11:
I performed a more investigations on this issue. The default symbol in the library is correct and usual MOSFET is also affected. The origin of this issue is related to the way how Qt processes the line thickness scaling. If you will zoom in the schematic, you will see the correct symbol. The Qucs uses a custom widget to render the schematic. If the schematic would be ported to QGraphicsScene, it probably would to fix this issue. Another way is to fix the MOSFET symbol in `mosfet.cpp`
[image]

tomhajjar, 2022-06-11:
I understand. Still odd the symbols look OK in the left panel menus and when dragging onto the schematic. They look wrong once placed.

ra3xdh, 2022-06-11:
Left panel uses pre-rendered PNG icons and therefore doesn't depend on Qt scaling. The symbol outline is rendered using line width=1 while dragging. The final symbol view uses thick lines that cause Qt rendering issues.

tomhajjar, 2022-06-11:
Is there a reason the choices for MOSFET symbols is so limited? There isn't a choice for enhancement mode devices. It makes schematics look odd when you have to choose the symbol for a depletion mode device with substrate for a common power MOSFET. I realize there is the Qucs legacy choices but they don't make it easy to enter Spice parameters.
 
[image]

ra3xdh, 2022-06-11:
I cannot say why the SPICE MOSFET symbols (dark red) were designed as depletion mode MOSFET. Probably it was simpler to draw one line instead of three lines. Concerning the parameters input, you may enter the model in the `Line_2`, `Line_3` parameters and use continuation line with `+` sign. Also you can put the model file to the `$HOME/.qucs/user_lib` directory and then drag'n'drop the model using the library manager. The `.MODEL` component will be created automatically in a such case. You need to reference it in the SPICE device.

ra3xdh, 2023-02-13:
I have fixed MOSFET symbols look. The fix will be available since v1.0.2 release. 
[image]

closed 2023-02-13 by ra3xdh