ra3xdh/qucs_s #94: Add SPICE models for transmission lines devices

ra3xdh, 2022-07-01:
The S-parameter simulation support for Ngspice and Xyce backends has been added in the 0.0.24 release. But the SPICE entries for transmission line devices including microstrip still be missing. These devices are unique for Qucsator and have no direct SPICE equivalent. It's need to provide a way to simulate old Qucs RF-circuits with new Ngspice supporting S-parameter analysis.

wsteffe, 2023-06-25:
From ngspice user manual (section 6.2.1 Lossy Transmission Line Model (LTRA)) I have seen that this simulator supports trnsmission lines and I have also seen that qucs-s provides a circuit element of this kind.

The only problem is that input data (O=LTRL specification) is not user friendly. I would expand it. 
In example:

---Input data
  Z=
  epr_eff=
  len=
  alpha=(Nepers/length) 

----  created ngspice model:
 //RLC (uniform transmission line with series loss only):
  G=0
  Vp=c/sqrt(epsr_eff)=1/sqrt(LC) 
  C=Z/Vp
  L=Z^2 C
  R=2 alpha Z

wsteffe, 2023-06-25:
Evren better:

--input:
alpha_dB=(dB/length) = conductorLoss/L computed in Tool/Line Calculator

---- created ngspice model:
R=2 (alpha_dB/20*ln(10)) Z

wsteffe, 2023-06-26:
May I ask to add also an ideal Coupled Transmission Line such as that one of ADS shown below?.  
It should be very easy to implement it using ngspice Coupled Conductor Line Model (see at https://nmg.gitlab.io/ngspice-manual/transmissionlines/kspicelossytransmissionlines/coupledmulticonductorline_cpl.html)
[image]

tomhajjar, 2023-08-15:
Not sure you saw Holger's last ngspice update on 2023-08-04 that includes built in S-parameter file support.

https://sourceforge.net/p/ngspice/discussion/120973/thread/51228e0b01/?page=3

ra3xdh, 2023-08-15:
Yes, I am aware that XFER device is accepted for Ngspice master branch. But I didn't start its implementation in Qucs-S yet. I decided to postpone the s2p files support to the next release. The upcoming `v2.0.0` release will be dedicated to the quick switching of the simulation kernels feature #274

ra3xdh, 2023-10-28:
Microstrip devices for Ngspice will be implemented as Verilog-A models. It's need to add the the support of system Verilog-A library first in order to resolve this issue. See #343

ra3xdh, 2024-02-14:
#564 adds support for RLCG device.

tomhajjar, 2024-02-17:
qucsator_rf will fill the gap. Most people using distributed matching elements are using active devices that only have s-parameters available.

I wouldn't want to design a narrow-band microwave filter using models based on Verilog-A models that have not been fully vetted by making and testing PCBs.

zergud, 2024-02-17:
I think we should provide a convenient mechanism to use qucsator-rf subcircuits (such microstrip filters etc) in ngspice simulation through s-parameters

ra3xdh, 2024-02-17:
@zergud  I am not sure if hybrid simulation it is needed for Ngspice. And I still don't understand its advantage. Everything is clear with OpenEMS/Qucsator hybrid simulation when OpenEMS exports s2p data. Such simulation would be useful if implemented. But I suspect Qucsator/Ngspice hybrid simulation will bring nothing except convergence errors.

ra3xdh, 2024-02-17:
I am more inclined to cancel the MS lines models task for Ngspice because OpenVAF is suspended and restrict MS simulation to Qucsator only.

zergud, 2024-02-17:
IMHO, it’s too early to talk about suspending of OpenVAF
anyway, this issue can be closed

tomhajjar, 2024-02-17:
Using Qucsator is a low risk solution. There are enough "Qucs" microstrip models to satisfy the average user. A large company is going to pay for ADS.

@zergud 

In general you won't have Spice models for most of these RF devices. One can always use transmission Spice models

zergud, 2024-02-17:
> In general you won't have Spice models for most of these RF devices. One can always use transmission Spice models

that's what I meant

ra3xdh, 2024-02-17:
I suspect the UHF amplifier like shown below will not converge using the existing s2p device implementation and Ngspice. Need to export s2p data manually and test. This schematic is using only two types of microstrips except cross. 

[image]

ra3xdh, 2024-02-17:
I decided to cancel implementation of native microstrip models for Ngspice and close this issue. May be reopened if some new ideas will appear in the future. But let's concentrate now on reverting Qucsator in the Qucs-S distribution #370.

closed 2024-02-17 by ra3xdh

tomhajjar, 2024-02-17:
The cross is not needed and there is only one transmission line type used. 

A simple Tline model that only uses Zo, length (deg) and F would work for most analysis.

tomhajjar, 2024-02-20:
See https://github.com/ra3xdh/qucs_s/discussions/601. I have an alternate way of replacing TLines in RF analysis.