ra3xdh/qucs_s #95: Navigation from Main Dock

tomhajjar, 2022-07-01:
Qucs 0.0.20 has the ability to navigate to a project in any fold
[image]
er from the Main Dock. I don't see that available in Qucs-S.

tomhajjar, 2022-07-01:
Qucs 0.0.20 has the ability to navigate to a project in any folder from the Main Dock. I don't see that available in Qucs-S.

ra3xdh, 2022-07-01:
Yes, I confirm that this feature is missing in Qucs-S. Only the directories following the `*_prj` template are enumerated in the left dock. It may probably be Qt5 limitation of the GUI control used for the left dock. I very rarely use projects for my personal simulations with Qucs-S, so I may overlook this feature.

ra3xdh, 2022-07-05:
Fixed by #100 , the advanced project tree will available in the next release.

closed 2022-07-05 by ra3xdh

ra3xdh, 2022-07-05:
The attached screenshot illustrates how the fixed project view look like
[image]